richonguzman/LoRa_APRS_iGate #210: Is it possible to set a different TX and RX frequency as Digirepeater

klslowik, 2024-12-30:
In Poland, in many regions we have adopted contractual frequencies at two different values. The so-called EU frequency and the PL frequency.
Trackers with EU frequency are relatively few. The problem begins when someone from outside Poland arrives and has the EU frequency set. Then, unfortunately, it is poorly captured in many parts of our country.
Therefore, for such people I would like to set up a Digirepeater, which would listen for packets on the EU frequency and retransmit them to PL.
Is there such a possibility?

EU
"frequency": 433775000,
"spreadingFactor": 12,
"signalBandwidth": 125000,
"codingRate4": 5,
"power": 20

PL
"frequency": 434855000,
"spreadingFactor": 9,
"signalBandwidth": 125000,
"codingRate4": 7,
"power": 20

richonguzman, 2024-12-30:
Hi,

digirepeater can work with no problem in a cross frequency, but i never needed to add the cross CR and SF.

maybe its time now to add it to make a full cross freq digi.

give me a few days to return home from holidays and i will try to add this

klslowik, 2024-12-30:
Hello
thank you very much for your quick response and your willingness to develop your wonderful software. If I could help you in any way you can count on me.
At the moment on my own also one of the versions of the software for the tracker I have developed so that with the SmartBeacon option turned off it broadcasts a packet at periods of fixed time alternating between PL and EU frequencies. This way, when driving in Poland, I don't have to manually switch the tracker. I am happy to send these changes to you as well.

Klaudiusz
SP9KSK

richonguzman, 2026-09-30:
for a while this has been solved, so I'm closing this

closed 2026-09-30 by richonguzman