richonguzman/LoRa_APRS_iGate #212: Invalid TX packet

madpsy, 2025-01-01:
I see lots of messages like this:

```<165>1 - IW2OGY-10 CA2RXU_LoRa_iGate_1.3 - - - TX / IW2OGY-10>APLRG1,WIDE1-1:}IW2NBW>APRS,TCPIP,IW2OGY-10*:;EL-IW2NBW*111111z4528.08NE00849.83E0144.638MHz T071 R16k SvxLink BOFFALORA TICINO```

That packet is invalid  because the `}` in this part `WIDE1-1:}IW2NBW` is errornous. I don't know if that's a logging bug or if that's what it actually tries to TX. Removing the `}` makes it a valid packet.

richonguzman, 2025-01-01:
This is Third party packet

When a packets comes from aprs servers

madpsy, 2025-01-01:
I wonder how it could make it through given it's not parseable, or at least the library I'm using can't. I'm fairly confident the `}` is being inserted by "something" incorrectly, just no idea where or how it could from APRS-IS in that bad state

wb2osz, 2025-01-02:
madspy -

The } data type indicator represents a third party packet.  When a packet is forwarded, from APRS-IS, to RF, the packet is encapsulated within a third party packet.  You might find it useful to read "Understanding APRS Packets" in https://github.com/wb2osz/aprsspec .

More details are in Chapter 17, of the APRS protocol spec **1.2**,  Don't use APRS101.pdf.  It is obsolete.

The processing of this type of packet depends on the type of station that receives it.

(1)	A digipeater only looks at the first via path (before the “}“) and doesn’t need to have any knowledge of third party packets.

(2)	An application that wants to interpret the information part of the packet must first remove the encapsulation and process what is left over.

(3)	An RF-to-IS IGate first removes the encapsulation and operates on the rest.  If the remaining via path contains TCPIP, the packet must not be forwarded.  This prevents loops.

Don't hesitate to contact me directly if you have any questions.

73,
John WB2OSZ
<callsign><at>arrl.net

madpsy, 2025-01-02:
@wb2osz Thank you very much for the explanation and pointers (also, great references you have in your repo!).

The issue appears to be that aprslib (any version) does not know how to parse these types of packets. By admission of the author the decode function is incomplete so this is simply and example of that.

I don't really need to parse these particular packet types so for now I'll skip them, though maybe tempted to patch the library in the future.

richonguzman, 2025-01-02:
> @wb2osz Thank you very much for the explanation and pointers (also, great references you have in your repo!).
>
> The issue appears to be that aprslib (any version) does not know how to parse these [...]

maybe you can use the un-encapsulated third-party packet for decoding.

this packets are coming from APRS-IS/Internet and Tx over RF , so the first part shows where it gets Tx and by whom.... but the real "original" packet is after this.

richonguzman, 2025-01-02:
closing this as its been explained :)

closed 2025-01-02 by richonguzman