richonguzman/LoRa_APRS_iGate #213: Feature Request: Add Minutes to GMT Time Correction Option

VO1BIL, 2025-01-02:
This a very very low priority feature request.  At some point, could you please include 30 or 15 minute intervals to the GMT Time Correction for accurate Local Time option? My local time zone is -3.5h from GMT.  Thanks for all your work, it is much appreciated.

richonguzman, 2025-01-02:
hi, 

did not know there was 0.5 time offset for timezone anywhere

it could be added easy, maybe you can flash firmware from github with a minor mods in /data_embed/index.html file to allow 0.5 time offset to test?

wb2osz, 2025-01-02:
It gets worse.   
There are places that are off by an odd multiple of a quarter hour.
https://en.wikipedia.org/wiki/List_of_UTC_offsets 

73,
John WB2OSZ

richonguzman, 2025-01-02:
> It gets worse. There are places that are off by an odd multiple of a quarter hour. https://en.wikipedia.org/wiki/List_of_UTC_offsets
> 
> 73, John WB2OSZ

well well well

i will add the offset for any multiple of quarter hour then :P

thanks for the info! You never stop learning

VO1BIL, 2025-01-02:
No problem, I am a computer tech and come across a lot of software that do
not include the 15 or 30 min time zone offsets, so I am use to making due :)

Yes I would love to test it, when you have time to make the changes.  I
really appreciate you taking time to look at it.

VO1BIL, 2025-01-04:
> please download github .zip latest update (not webflasher) to test on your
> side (only firmware update is needed)

I installed the latest firmware as you suggested. I now have the option to
change and save a -3.5h offset.  However when I click save, it reverts to
-3h.

richonguzman, 2025-01-04:
found the issue, it should be ready for testing (working on my iGate)

VO1BIL, 2025-01-04:
Yes it works for me now.  Thank you!

richonguzman, 2025-01-04:
thanks for the coffee ;)

closed 2025-01-04 by richonguzman