richonguzman/LoRa_APRS_iGate #215: When telem

aymericmo, 2025-01-03:


richonguzman, 2025-01-03:
hi,

I confirmed with the developer of aprs.fi and many others that WX packet blocks the decoding of any encoded telemetry added in the comment of the packet.

so you are right:
its a Wx packet and voltage is sended as regular text
or, its a normal gps position beacon with encoded telemetry

the only solution I can see for this is to send (when Wx packet is selected) a new normal gps pack at a different rate to avoid losing the battery telemetry

syl87, 2025-01-03:
Hi All
Would it be possible to use an automatic altern telemesure beetween voltage and weather when they are two enabled?
73
Syl F1UJT

richonguzman, 2025-01-03:
> Hi All Would it be possible to use an automatic altern telemesure beetween voltage and weather when they are two enabled? 73 Syl F1UJT

this could be a good solution... like sending it once 30 min or 1hr?

or just alternating one or the other at the "normal" 15min packet rate?

aymericmo, 2025-01-03:
I think normal packet rate 15 min is fine, wx sensors 15 min after voltage and add a comment in WebGUI to explain this mode.

syl87, 2025-01-03:
There is two way you can use. A semaphore to switch at each beacon voltage/wx  for a time choosen by the user, or having 2 independant routines to send this alternatively (WX at 0 and Voltage at 0 + half of user time ?)

SQ2CPA, 2025-01-03:
> Hi All Would it be possible to use an automatic altern telemesure beetween voltage and weather when they are two enabled? 73 Syl F1UJT

I have something like that running now for testing at `SQ2CPA-13`, you can check raw packets.

aymericmo, 2025-01-03:
> I have something like that running now for testing at `SQ2CPA-13`, you can check raw packets.

It's seems good ! You send telemetry and wx alternatively ?

SQ2CPA, 2025-01-03:
As you can see there are two type of frames:

1. Encoded telemetry (with L# symbol)
`SQ2CPA-13>APLRG1,WIDE1-1,Q1251,qAO,SQ2CPA-4:!L3Q7^Rw`q#  GLoRa APRS PV WX test|'I%Y!.|`

2. WX data (with L_ symbol)
`SQ2CPA-13>APLRG1,WIDE1-1,Q1352,qAO,SQ2CPA-4:!L3Q7^Rw`q_  G.../...g...t072h34b09864LoRa APRS PV WX test`

I have interval set to 15 minutes and I rotate the frames so:
WX frame every 30 minutes
Telemetry frame every 30 minutes

It works fine, I think that is the best method to have WX gate with Telemetry and WX at once. I don't want to send 2 frames one by one because that's implemented in low power mode so I want to save power and I also don't need WX data and telemetry that often

and one important thing - it only works that way with encoded telemetry and WX, if you have "telemetry" after comment then it works normal with WX and voltages in comment. 

I also wanted to use the same L_ icon for encoded telemetry frame but it doesn't work (doesn't parse the data at all) so I have to use standard L#