richonguzman/LoRa_APRS_iGate #216: A lot of CRC ERROR

cvaldess, 2025-01-05:
Hi,
Running iGate Version date: 2025.01.02 in LILYGO TTGO T-Beam V1.2 and when check Syslog see to many crc error.

Jan  5 01:14:34 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: TX / EA4GLO-10>APLRG1,WIDE1-1:!L:*-2MSQX_  G.../...g...t054h55b09487LoRa APRS iGate TTGO T-Beam V1.2 Batt=4.17V
Jan  5 01:15:06 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EC4TR-10>APLOX1,IDE1-1:!/9s>uMeZRy  XEC4TR Lora iGate / -128dBm / -19.50dB / 3461Hz
Jan  5 01:16:15 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4HXE-10>APLRWa}▒▒|^▒▒y
?dyt▒mQ▒▒▒\▒▒
             ▒▒tf<#Av▒a▒▒\r▒b)
▒:▒@ / -125dBm / -17.75dB / 3479Hz
Jan  5 01:16:18 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4ATS-10nWGq<D4ZAH-4*:!L
 HjNPrVa  GLoRa APRS / -130dBm / -21.50dB / 140Hz
PuTTYPuTTYJan  5 01:17:13 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4DD]<!▒▒M▒▒O▒▒▒\▒0ZAH-4*:!L▒#▒pDGYz&  XEA4DTE iGate ▒%'' P=4.43V / -129dBm / -21.75dB / 147Hz
Jan  5 01:17:31 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4AC▒▒▒4KPLRG1,ED4ZAH):=4017.82NL00359.30W_...OH.(▒..▒60r...p...P...h%$c18▒7<Lo^a APRS / -129dBm / -19.75dB / 198Hz
Jan  5 01:17:52 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4D-11>APLOX1▒▒▒\R%▒8▒WIDE1▒▒+▒,sjMX!Na  XEA4D Lora iGate System P=4.37V / -130dBm / -19.50dB / 3488Hz
Jan  5 01:18:23 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: RX / GPS / EA4SG-10 / APLRG1 / WIDE1-1 / -121dBm / -12.25dB / 2129Hz / 40.65000N / -3.79001E / 4537.6km / LoRa APRS iGate en TTGO. 73 de David|*:%d|
Jan  5 01:18:27 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒▒▒▒G-10>APLRG1,ED4ZAH-4*:!L9w**MQlP&  GLoRa APRS iGat▒ i▒ X▒GG▒ 73 de David|*:%dx / -129dBm / -21.25dB / 192Hz
Jan  5 01:18:33 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4SG-10>APLRG1,EC4R(10,WIDE1*:!L9w**MQlP&  GLoRa APRS iGate en TTGO. 73 de David|*:%d| / -128dBm / -19.25dB / 3499Hz
Jan  5 01:20:26 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4GWT-10>BPLRG1,ED4z▒=4:;Mf▒4GEN>PZKWF,TCPIP,E▒$▒OE$10*:;EA4GMF_1 *050▒20z<026.61N/00322.12WyAirSpy EA4GMF / -129dBm / -21.75dB / 141Hz
Jan  5 01:22:09 EA4GLO-10 CA2RXU_LoRa_iGate_1.3: CRC / CRC-ERROR / <▒EA4EP▒11>APLRG1,ED4ZAH-4*:!L:$Y^MSM+a xGLoRa APR / -129dBm / -21.00dB / 137Hz

richonguzman, 2025-01-05:
Maybe there are real CRC errors

Check the packet Rx to see what's going on

cvaldess, 2025-01-05:
list the 10 last received packets, here the last four

00:45:09	EC4TR-10>APLOX1,WIDE1-1:!/9s>uMUYQy XEC4TR Lora iGate	-129	-20
00:48:22	EA4SG-10>APLRG1,WIDE1-1:!L9w**MQlP& GLoRa APRS iGate en TTGO. 73 de David|*<&.|	-120	-12
00:56:15	EA4ATS-10>APLRG1,WIDE1-1:!L:#xYMSrVa GLoRa APRS	-126	-16.75
00:56:46	EA4GLO-7>APLRT1,WIDE1-1:!/:*-?MSQ#[ pQ	-64	11.5

but in logs show a lots of crc errors

richonguzman, 2025-01-05:
Received packets are all the non-crc errors packets

So it's normal to only see CRC error listed in the syslog log and not on the webUI

aymericmo, 2025-01-06:
> -128dBm / -19.50dB

All of packets are weak signal / SNR.

richonguzman, 2025-01-06:
> > -128dBm / -19.50dB
> 
> All of packets are weak signal / SNR.

exactly, the lower the RSSI/SNR the more likely to get CRC

remember: those values are way below the noise

where is this station? real high listening to lots of packets?

cvaldess, 2025-01-09:
was checking my antenna installation and not doing good at 433,  now running on Vertical VHF/UHF, will change it soon

richonguzman, 2025-01-09:
should I close this issue now?

cvaldess, 2025-01-11:
Yes, please

closed 2025-01-11 by richonguzman