richonguzman/LoRa_APRS_iGate #217: too fast(?) reboot after restoring from backup

lemielek, 2025-01-05:
I did backup of my TTGO and flash RDZ_sonde fw to it to check oled display work.
After test I flashed igate fw again and tried to restore settings.
After Save in browser there was no Save confirmation in browser. 
Save confirmation - succes is showed after reboot.
But from ttgo is served not changed (with restred data) data, but from full flash (nocall-10, etc.)
Maybe it is to much data to save and 2 seconds or less is not enogh before reboot.
Saving only AP and password is succesfull. 

Maybe there should be some wait?
https://github.com/richonguzman/LoRa_APRS_iGate/blob/1ad13e3c09368cca0b4d7b6928254409b54af01f/src/web_utils.cpp#L225 

Pic from Chrome DevTools
[image]

richonguzman, 2025-01-07:
when you save, WIFIAP is killed and no web confirmation will appear.
did it work ok after saving the data?

lemielek, 2025-01-07:
Works, but save was no succesfull, no change in data, still old NOCALL instead my SQ9CWN, etc.

richonguzman, 2025-01-07:
could you flash it again with latest, erase all , and then only change your callsign and save to test?

lemielek, 2025-01-07:
Flashed v2.2.1
Call changed and saved succesesfully.
But after reconnection tried again importing backup json file, saving and no success, still only call changed like before restore try.
The same was when I was only adding my WiFi data after flashing.

But. I set all manually, did additional backup and flash with erase again, and from this fresh backup all was restored succesfully.

richonguzman, 2025-01-07:
so it may be that your "old" restoring json file did not have all that was needed in the new "updated" version?

I've seen this bug sometimes when I add a few new things to the webUI to control more things

please confirm that adding the data manually did work

lemielek, 2025-01-07:
Yes

closed 2025-01-07 by lemielek