richonguzman/LoRa_APRS_iGate #219: Igate/Repeater in repeater ony mode does not send beacon

SQ2CPA, 2025-01-06:
As @Jan-OK2 said here: https://github.com/richonguzman/LoRa_APRS_iGate/discussions/218

"
Hello.
I am new to use LoRa for APRS platform and I am testing amazing projects created by CA2RXU. I need to build cross frequency repeater, which besides digipeating send beacon with WX data from BME 280. In its location isn't electricity and internet. When I use it as digipeater without APRS-IS connection, it repeat good but not send beacon. I have not find reason why. For me is very important to monitor baterry voltage and useful to monitor WX. Can You help me?
"

I think I have the same problem. In the newest firmware I can see only "status" packet after boot and no beacon packet (that should be send after booting without any delay). Also "send beacon" in web ui doesn't work and it also should send beacon after clicking (that helps testing some stuff)

I have APRSIS connection disabled. I think it will also don't work at fresh config without any changes.

Damian SQ2CPA

richonguzman, 2025-01-06:
this was corrected on v2.2 as when changing code for aprs-passcode verification process was added I broke the "sendBeacon"

but now is fixed... please update

closed 2025-01-06 by richonguzman

SQ2CPA, 2025-01-06:
ok but I found

https://github.com/richonguzman/LoRa_APRS_iGate/blob/1ad13e3c09368cca0b4d7b6928254409b54af01f/src/utils.cpp#L189

This condition is invalid. Beacon is beacon and it have nothing to do with digi or APRSIS. For example I can use station for TNC without APRSIS and I also want to use beacon from this station to send my home position via RF and I don't want to repeat any frames. 

This beacon interval should always work or only if `sendViaAPRSIS` and/or `sendViaRF` enabled.

richonguzman, 2025-01-06:
read it again...

if igate, checks aprspasscode
if digi, sends it without this verification

so its pretty accurate

SQ2CPA, 2025-01-06:
beacon should not depend on anything except `sendViaAPRSIS` and `sendViaRF`