richonguzman/LoRa_APRS_iGate #221: Heltec Lora32 new revision V3.2

OM2RC, 2025-01-08:
Hello dear Ricardo, Aliexpress seller has sent me Heltec board in new revision V3.2 (HTIT-WB32LAF) . I have tried to use the latest V2.2.1 iGate Heltec V3 code through web flasher, but OLED display not working. I am able to connect to AP wifi, but after my ssid/passcode configuration, it is not able to connect to my wifi network (switched back to AP). LoRa module seems working correct, RF packet is sent OK. Any idea what to do, or try ? Thank you ! 73

richonguzman, 2025-01-09:
Hi, i will have soon a beta test firmware for this new board as it seems to change a bit and dont work with old version of Heltec V3

richonguzman, 2025-01-09:
with the help of another ham we found out what difference has this board (V3.2) with the regular (now old) V3.

so new adjustment are coming for this board soon (tonight or tomorrow) wait a little

just check if new release is available and then webflasher should reflect this right away

richonguzman, 2025-01-10:
hi, can you test the latest branch on github "HELTEC_V3_2" to see if this works for you?

OM2RC, 2025-01-10:
hi, sorry, I am not experienced enough how to do it - I have successfuly created "Build" from your platformio, but during upload I have obtained this error message: A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument? I see environment: ttgo-lora32-v21  with status: FAILED.  But maybe I should change something before, or do something different (?)

richonguzman, 2025-01-10:
to upload from VSCODE, you need to change the ENVIRONMENT in VSCODE after importing the firmware.

To do this go on the bottom part of the window, where it says "DEFAULT t-lora32-v21" or something like this....  press with the mouse once, a new window will open after a few secs and there you select HELTEC_lora_wifi_V3_2 or something like this

OM2RC, 2025-01-10:
ahh, ok, thank you ! Now it seems much better. Display OK, wifi OK, RF OK. Good job Ricardo, thank you !
[image]

richonguzman, 2025-01-10:
great news

can you test/confirm that it gets good reading of the battery voltage too?

OM2RC, 2025-01-10:
ok, I will connect battery power and compare voltage measurement, but probably tomorrow, during the day...

OM2RC, 2025-01-12:
> ok, I will connect battery power and compare voltage measurement, but probably tomorrow, during the day...

Sorry for delay. My measurement showed 6 mV difference in 4 different voltage points.

closed 2025-02-25 by richonguzman