richonguzman/LoRa_APRS_iGate #227: Heltec V3.2

umj, 2025-01-13:
Hello Ricardo

I just received two Heltec V3 They would not load on the Heltec V3 and the flash caused it to loose display , It also did the same on the Lilygo T3S3 . The Lilygo is still down

I did get the Heltec to load using the Heltec V3.2 flash and the display did return.

I connected to the configuration page and entered my configuration added my network and all seemed to go well until I hit save, 
The Heltec display went black and back on again as if it had loaded and rebooted, The Display Read Lora APRS  and some other information 

I've attached the display  

Thank You and 73
[image]

richonguzman, 2025-01-13:
Hi, can you flash it again (deleting all data) and save only your Callsign and one Wifi SSID + Password (to test if this runs well)?

umj, 2025-01-13:
I tried it and it took only the information I entered, I was able to login to the IP address, But when I added anything else on the configuration page it went back to the previous display that I attached and won't connect to the ip address that I entered, I'm back to only callsign and wifi ssis and password now...  next step please!
[image]

richonguzman, 2025-01-13:
what are the next mods you did?  APRS-IS on and also APRS passcode?

(avoid any "\" on any text box)

maybe describe all that you need to activate to see where it may be the error

richonguzman, 2025-01-13:
can you post the APRS-IS config you were trying? (this is because you mentioned all else worked fine, right?)

umj, 2025-01-13:
[n9umj-4_LoRa_iGateConfigurationBackup.json](https://github.com/user-attachments/files/18400772/n9umj-4_LoRa_iGateConfigurationBackup.json)

This is what I was able to configure and get to work without any errors

Thank You for your assistance

Rick

richonguzman, 2025-01-13:
it seems on the .json file that all has been activated and running, right?

or am I missing something?

umj, 2025-01-13:
Yes it is running, It locks up and gives me the first display when I try to change the frequency from 433 mhz to 916 mhz
How or where do I select and put the tx and rx frequency for the i-gate?

I was able to enter all that you see in the config file, I got to the frequency config and it took me back to the original display lockup!

Rick

richonguzman, 2025-01-13:
so, what are the values you are entering for Freq?

are you changing both freqs to the same 916000000 ???

umj, 2025-01-13:
Yes, I'm trying to set them to 91650000

richonguzman, 2025-01-13:
> Yes, I'm trying to set them to 91650000

"91650000" is only 91MHz , you are missing a ZERO, right?

umj, 2025-01-14:
Ricardo....Yes ..Your Right! I was missing a 0, .....Working fine Now!

I appreciate your help. and all you've done with the project.....I do remember where the paypal button was located!!

73
Rick   -n9umj

richonguzman, 2025-01-14:
glad you solve the issue finally!!! now enjoy ! 73!

closed 2025-02-25 by richonguzman