richonguzman/LoRa_APRS_iGate #229: APRS raw packets display and battery percentage stuck.

iuliandj, 2025-01-14:
Hello.
Can you please add in the next firmware for i-gate the posibility to show raw packets data and text in the comment? 
I use the tracker as a network packet monitoring also, and I can not see the text messages sent in comment or beacons packets, and also in status packets.
In menu no.3 Stations -> Packet Decoder, I can only see on display the callsign, position and RSSI, but I am interested in the transmitted text putted in comment of each APRS station, or beacons messages,  and the posibility to show path of the received packets.

I notice a bug in last firmware from 11.01.2025, on display it remains stuck 91% battery percentage. The value is modifying but percentage not. This happened after a complete discharge of battery and charge it to 91%, after that I decoupled from charger because of need. 

Thank you verry much!

73 from YO7MGH

richonguzman, 2025-01-14:
Hi,

thanks for donation !

as for the required updates:
a) for igate you would want to see the decoded data for each packet instead of the basic data seen right now?
b) for the tracker also the same? all data decoded instead of basic data?

iuliandj, 2025-01-14:
My mistake. I need to see the data and text in tracker software. I gate is ok.

iuliandj, 2025-01-14:
Yes. I updated my last comment with what I was reffering. A modification to Packet decoder to see text also, and a sub menu Raw data where it will be showed all pakets received with text and path.

richonguzman, 2025-01-14:
> Yes. I updated my last comment with what I was reffering. A modification to Packet decoder to see text also, and a sub menu Raw data where it will be showed all pakets received with text and path.

got it, will try to add this as soon as I finish adding a new special board

iuliandj, 2025-01-14:
Ok, thank you!
Please check also the situation with 91% stuck on display if there is any solution, on tracker firmware.
 The battery is charghed to 4 volt but it keeps showing 91% on display, even if is below 3 volt. In the first hours of working with new firmware was ok, uitill first total discharge if battery. 
 
Regards,

iuliandj, 2025-01-18:
Good morning. 
In I-gate firmware, at APRS-IS Filter section, multiple filters does not work. It works only first one. On T2 server that is connected to, it appears entire row but on air it sends only first command. I read that multiple filters can apply by putting space between commands, but it dos not what to work.
Also if I put filter b/call1/call2… it takes from IG to RF only packets from call1.

Thank you!

Mane76, 2025-01-18:
Hello iuliandj, just to get you right, 

- you put the filter command in the respective line in igate
- the T2 server shows the filter command correctly

if the packet you receive from aprsis is a message the igate will only tx if call1 and call2 was heard before in the last 30mins
if the packet you receive from aprsis is an object you need to enable to tx objects to RF

Do both above lines match your situation ?

Why do you use such a strange filter command at all ? The result will be a TX capable iGate which will only TX packets from call 1 and call 2.

BR
Mane

iuliandj, 2025-01-18:
> Hello iuliandj, just to get you right,
>
> * you put the filter command in the respective line in igate
> * the T2 server shows the filter command correctly
>
> if the packet you receive from aprsis is a [...]

Hi.
Igate sent to RF only call1 when it receives a packet on the aprs server.
Call2 sent position packet to IG but IG does not sent to RF the packet, in my situation.  In web GUI messages and objects are enabled, of course.
I guess that positions packets and any comments or status packets, telemetry are not defined to pass from IG to RF.

Ricardo can help us here figure it out.

richonguzman, 2025-01-18:
how much distance between Stations2 and iGate?

if distance is not good defined in filter it could also not be Tx info Lora-RF
did station2 send gps beacon or just status/messages in the last 30 min?

iuliandj, 2025-01-18:
According to filter description “b/call1/call2...	Budlist filter	Pass all traffic from exact call: call1, call2, ... (* wild card allowed)” distance is not needed.

Call2 sent a position beacon with text but there was not IG to RF with call2, call3 etc. I tried with “ / “between calls, with space between them, with “ , “ with “b” in front of esch call and nothing. Only “b/call1” works.

richonguzman, 2025-01-19:
can you share "call2" real callsign?

the igate has a Reporting distance that works on real GPS distance and not filtering distance at all

iuliandj, 2025-01-19:
I tried with different callsigns.
You can test with YO7MGH-10 and YO2B-10

richonguzman, 2025-01-20:
so:
YO7MGH-10 is listened and also can forward from APRS to RF to this station.
YO2B-10 is not listened and not forward any packet to it?

can you share the distances between YO7MGH-10, YO2B-10 and igate?

iuliandj, 2025-01-20:
MGH10 is about 120 km and B10 is about 140 km.
Both stations are not forwarding from IS to RF when they are on call2 in filter.

richonguzman, 2025-01-20:
And the distance defined on the igate web config?

iuliandj, 2025-01-21:
No distance defined because if we look at filter description we put only command and calls:

“b/call1/call2... Budlist filter Pass all traffic from exact call: call1, call2, ... (* wild card allowed)”

Mane76, 2025-01-21:
Please understand that there are two chains working:

1) With filter set you just tell aprs-server what shall be sent from APRSIS to iGate
2) With distance set in iGate and Message/Object slider the iGate decides what to send to RF and what not

Both conditions must be met that a frame is sent via RF

iuliandj, 2025-01-21:
> Please understand that there are two chains working:
>
> 1. With filter set you just tell aprs-server what shall be sent from APRSIS to iGate
> 2. With distance set in iGate and Message/Object slider the [...]

I don’t see that information here, in the official link.
http://www.aprs-is.net/javAPRSFilter.aspx

Applying your info, how you will define b/call1/call2… command please ?

Mane76, 2025-01-21:
Please understand that you mix up two things.

- The link you provided is the general server side APRS filter commands every APRS-Server understands. You can use this filters to define what data should be sent from APRSIS to a client. A client can be Ricardos Software as an iGate but can also be a terminal in a telnet-session to the APRS-Server.
- Condition 2 in my answer is a thing how Ricardos Software does handle data to be sent to RF. We have the opinion that stupid sending everything makes no sense. Therefore we implemented two conditions. Distance and time. If a packet is coming from APRS-Server to iGate the iGate checks a) did I hear that station in the past 30mins (parameter rememeber_station_time) and is it withing the range of x km (parameter m/10 in the aprs-filter command). Only if a) and b) is met the iGate send the packet to RF.

To answer your question I need to understand your usecase. Could you please explain what in detail you want to reach with your TX capable iGate ?

iuliandj, 2025-01-21:
> * Condition 2 in my answer is a thing how Ricardos Software does handle data to be sent to RF. We have the opinion that stupid sending everything makes no sense. Therefore we implemented two [...]

I understand now why b/  filter command does not work. 
Is not ok for me to reply and explaine and after that for you to reply and another reply and so on and on and on ...

My request is that filter to be activated as it was originally defined. No distance required, and the posibility to put 4 to 6 calls.
I DO NOT consider this a STUPIDITY. 
In my opinion, as an user. if I have an APRS device, my expectations are to have all the functions available, as other aprs devices.
In this manner, the user can be delighted with the device and recommend it to others. It's not ok to block or restrict/limit some features because they appear as supid for some and good things for others.

Is Ricardo decision if he want to enable b/call1/call2/call3/call4... filter nor not.. 
That is my request and what I spooke and agreed with him at the start of this threat. 

Thank you !

Mane76, 2025-01-21:
Apologies if you got it wrong.
My intention was to explain you in detail what is happening at aprs-is and iGate Firmware, and of course **you can** send everyting to RF **if you set the parameters right** - therefore they are here and therefore it is not hardcoded.

I offered you support in setting the parameters right if you can explain me your usecase.
As I understand this is not your intention - also fine fore me.

Best regards

iuliandj, 2025-01-21:
> Apologies if you got it wrong. My intention was to explain you in detail what is happening at aprs-is and iGate Firmware, and of course **you can** send everyting to RF **if you set the parameters right** - therefore they are here and therefore it is not hardcoded.
> 
> I offered you support in setting the parameters right if you can explain me your usecase. As I understand this is not your intention - also fine for me.
> 
> Best regards

And once again we get to replies and explanations that were not necessary if the filters did not have an additional filter.

The usecase is this. 
I want to receive from IG to RF the following:
- in b/ filter, all packets that come from calls defined in filter, without distance limit
- in e/ filter, all packets with the specified callsign-SSID(s) immediately following the q construct. This allows filtering based on receiving IGate, without distance limit
- in d/ filter, all packets that have been digipeated by a particular station(s) (the station's call is in the path), without distance limit

This is it.

According to this https://www.aprs-is.net/javAPRSFilter.aspx, my needs are met if I use these filters. 
I put them in filter with no result. 
So, what parameters need to be putted and where ?

richonguzman, 2025-01-21:
Just validating (again) some stuff

Are both stations listened by the igate? Regularly?

iuliandj, 2025-01-21:
> Just validating (again) some stuff
> 
> Are both stations listened by the igate? Regularly?

No. They are not. Never. 
Is more than 120 km distance between them so that is why I need to pull them from IG to local RF when they transmit packets to IG.
Also stations that I test with e/call filter and d/call filter, are more than 200 km away and with no direct contact on RF by my IG.

richonguzman, 2025-01-21:
So why do you want to Tx to RF those packets?

iuliandj, 2025-01-21:
So I can have some traffic in Lora APRS because in my area is not, and because I want to monitor the packets send by those calls, on RF.

iuliandj, 2025-01-22:
@richonguzman 
I discover that on 11 january tracker firmware, Weather report is not working. You can sent query but with no response. I loaded vers from 21 december, that I know it works, and it worked. I receive the answer from CA2RXU-15. I tested in different days, gps connected.

richonguzman, 2025-01-22:
hi YO7MGH, 

there a 3 topics in this ISSUE REPORT.

can we do one for each problem?

iuliandj, 2025-01-22:
> hi YO7MGH,
> 
> there a 3 topics in this ISSUE REPORT.
> 
> can we do one for each problem?

Sure, you can do.

richonguzman, 2025-01-22:
please, post one for each one (and then I close this to go topic by topic)

iuliandj, 2025-01-22:
Done!
I created 4 topics, each one with her subject.

Thank you, good luck and apreciate yout work. 
Don't forget to check your paypal next week. :-)
You can close this.

Mane76, 2025-01-22:
> According to this https://www.aprs-is.net/javAPRSFilter.aspx, my needs are met if I use these filters. I put them in filter with no result. So, [@Mane76](https://github.com/Mane76) what parameters need to be putted and where ? Examples please. no more questions.

You can apply a m/500 before your set of filters, this would give the igate the distance parameter which is needed.
BUT
as you said your igate does not have heard one of the stations via RF it will not transmit. 
This behaviour is hardcoded - so no solution for your demand.

BR