richonguzman/LoRa_APRS_iGate #230: MESHCOM 4.0 MODE no PACKET RX

rocktester, 2025-01-16:
Hallo ....

 at FIRST... many thanks for your GREAT WORK and SOFTWARE!!

i tried to receive Packets in MESHCOM 4.0 Mode .....

uses SF11 BW 250KHz CODNIG RATE 6 but nothing was on on my local test

APRS in SF12 BW125 CODING RATE 5 woked fine.... 

MAYBE you can look on IT.....

https://icssw.org/en/grundlegende-spezifikationen/

THANK YOU VERY MUCH 
regards and greetings from Mike DG0OPK

richonguzman, 2025-01-16:
hi, do you want to hear and decode Meshcom FW on my FW?

with different CR4, SF and BD they wont communicate with each other. There should be at least the same Freq configuration for it. Then is Meshcom with AX-25 encoded data and sended as HEX values?

rocktester, 2025-01-17:
Hallo and sorry for delay.....
I want to use it as RX Monitor , to see whats coming out via RF of my Meshcom Modules.
Your Software has the possibility to lsn only,and free use of LORA parameters but it does not work with the LORA Meshcom parameters.
regards Mike dg0opk

rocktester, 2025-01-17:
i dont know if it has a special byte for starting but i could not receive anything from my Meshcom nodes.... ARPS decode workes well.
Maybe Kurt OE1KBC has the answer?
contact..... e-mail-adress:  oe1kbc@icssw.org
should i ask him?

regards Mike dg0opk

richonguzman, 2025-01-18:
please mike talk directly to him to let us know if he has one or few special bytes to get if this is the reason igate is not decoding MeshCom packets

rocktester, 2025-01-25:
found this in the source code....example for EU settings

        case 8:  // EU Preabble 8 ... 
            meshcom_settings.node_freq = RF_FREQUENCY;

            #if defined BOARD_RAK4630
                meshcom_settings.node_bw = 1;
                meshcom_settings.node_cr = 2;
            #else
                meshcom_settings.node_bw = 250.0;
                meshcom_settings.node_cr = 6;
            #endif

            meshcom_settings.node_sf = LORA_SF;

            meshcom_settings.node_track_freq = LORA_APRS_FREQUENCY;

            meshcom_settings.node_preamplebits = 8;

            break;
regards mike

rocktester, 2025-01-25:
correction .....the default LORA_PREAMBLE_LENGTH for EU is 32
 default:    // EU
            meshcom_settings.node_freq = RF_FREQUENCY;

            #if defined BOARD_RAK4630
                meshcom_settings.node_bw = 1;
                meshcom_settings.node_cr = 2;
            #else
                meshcom_settings.node_bw = 250.0;
                meshcom_settings.node_cr = 6;
            #endif

            meshcom_settings.node_sf = LORA_SF;

            meshcom_settings.node_track_freq = LORA_APRS_FREQUENCY;
            
            meshcom_settings.node_preamplebits = LORA_PREAMBLE_LENGTH;

richonguzman, 2025-01-29:
there seems a lot of differences in the way they transmit data and I have no info about the packet format they use. Will try to gather some info about this in the next week.