richonguzman/LoRa_APRS_iGate #233: freeze on lora power 1

9a3zi, 2025-01-18:
Hi, for info,  I just tried to change the power on the lora from 20 to 1 and after reboot not working, it cannot be connected to the web server and in the serial monitor there is only: 
Starting Station: 9A3ZI-10   Version: 2025.01.11
(DigiEcoMode: OFF)
Initializing LoRa Module
Starting LoRa failed! State: -13

It is about ESP32 + Ra-02 (DIY). Otherwise, with power 20, everything worked as it should. As far as I can see, only reprogramming helps now.

richonguzman, 2025-01-18:
For now power selection is dependent of the Lora module 

Soon will update the way power is selected to add more flexibility for this

Please remain in know working values

9a3zi, 2025-01-18:
> For now power selection is dependent of the Lora module
> 
> Soon will update the way power is selected to add more flexibility for this
> 
> Please remain in know working values

hi,  here I tried it, the TX works properly at 2-17 and 20. it freezes at power 1, 17 and 18. My hw: Ra-02 Ai thinker module is connected to the ESP32 wroom DIY i-gate. Programmed with web flasher for ESP32 + LoRa SX1278. I hope I helped. 73.

richonguzman, 2025-01-18:
great info, will soon have this in a dynamic way to reflect all the possibilities of each LoRa Module in each board