richonguzman/LoRa_APRS_iGate #234: TTGO T-A7670 E board - No Internet connection via LTE

DG3SBI, 2025-01-18:
Hi Ricardo,

great job and many thanks for this project.
I tried to establish a LoRa APRS iGate via mobile Internet Connection with a TTGO T-A7670 E R2.
The web flashing process works fine . 
But as soon as i switch on >Enable APRS-IS connection< , >SAVE< and reboot the device,
the boot process stucks as following (Version 2.2.2):
*CALLSIGN-xx*
Opening Port...

I think, it is due to the missing mobile internet connection . There is no PIN on my simcard...
In Germany it is mandantory to fill in the >APN< **and** the >username< for a mobile internet connection by your carrier e.g. Deutsche Telekom. Otherwise the mobile network refuses the Internet Access by the device.
A decription of the AT command is in the following datasheet: https://www.simcom.com/product/A7670X.html 
=> A76XX Series_AT_Command_Manual_V1.06.pdf

Would it be possible to provide such input fields for APN and username in one of the next releases of your software for the TTGO T A7670X ?

vy 73 de
Martin DG3SBI

richonguzman, 2025-01-18:
Hi

Sure I can add this ... are you willing to help me test this too?

This way we can create it faster 

I use telegram (same user as github)

DG3SBI, 2025-01-18:
Hi Ricardo,
of course i will support you by testing.
So far, i have no telegram account...
I am not so familiar with the classic ESP flashing process. But with a 
little assistance / description we will succeed.
Regards
Martin DG3SBI

------------------------------------------------------------------------
Gesendet mit der Telekom Mail App
<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

--- Original-Nachricht ---
Von: Ricardo Guzman (Richonguzman)
Betreff: Re: [richonguzman/LoRa_APRS_iGate] TTGO T-A7670 E board - No 
Internet connection via LTE (Issue #234)
Datum: 18. Januar 2025, 22:52
An: richonguzman/LoRa_APRS_iGate
Cc: DG3SBI, Author

Hi

Sure I can add this ... are you willing to help me test this too?

This way we can create it faster

I use telegram (same user as github)

richonguzman, 2025-01-18:
Any message app you use?

Or just plain emails?

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

<https://kommunikationsdienste.t-online.de/redirects/email_app_android_sendmail_footer>

DG3SBI, 2025-01-19:
Due to lots of spam and ID-spoofing, i reduced the use of message apps to an absolute minimum.
=> contact prefered by plain eMail.
Regards 
Martin DG3SBI

IZ1HMT, 2025-05-18:
Looking at other firmwares that had the same problem, you should insert this string: AT+CGDCONT=1,"IP","your.apn.here" somewhere that Ricardo might know.
I'm waiting for a TTgo with 4G too and I already know that my telephone provider needs to configure the "apn" so I will soon have the same problem too

richonguzman, 2025-05-18:
> Looking at other firmwares that had the same problem, you should insert this string: AT+CGDCONT=1,"IP","your.apn.here" somewhere that Ricardo might know. I'm waiting for a TTgo with 4G too and I already know that my telephone operator needs to configure the "apn" so I will soon have the same problem too

let me know when it arrives to start getting a way to solve this together

IZ1HMT, 2025-06-27:
The module has arrived. With various changes made thanks to ChatGPT I managed to insert the APN configuration of my telephone operator into the firmware. It also connects to aprs.fi but does not release any beacon or data. I think the firmware is not compatible with any A7670x with lora module.

NoahDoah, 2026-01-01:
Ok, so I actually got it to work now. First using ChatGPT, then Claude found two important things. And in the end I connected CODEX to VSCode which was epic and then it worked right-away.

Also make sure you got the RX and TX pin right. I'm a microcontroller beginner and seem to have mixed those pins up definition-wise. If you get timeouts when trying to query or connect the A7670 modem that might be the case.

I have no idea how clean the solution is, but it connects to APRS-IS and sends the beacon and also feeds tracker packets into APRS-IS. 

You only have to change the APN config in the complete code file here:

```
  const char* APN  = "internet.t-mobile.at";
  const char* USER = "";
  const char* PASS = "";
```

If anyone might need help on how to manually flash it to the ESP32 I'm happy to tell you, as I also just learned that a short time ago.

Edited code:
[A7670_utils.cpp](https://github.com/user-attachments/files/24399819/A7670_utils.cpp)

### Changes

1. There are two helper functions created that add a more robust modem connection and debug outputs (from what I've understood)
2. The APN consts have been added
3. Some changes in `void APRS_IS_connect()` for more debug and the connect including the APN
4. A7670 APRS-IS login responses (logresp, even with leading #) are now forwarded to processAPRSISPacket, so the APRS-IS verification line isn’t dropped.
5. When the modem reports “verified” during APRS-IS login, passcodeValid is set to true immediately, allowing beacons to upload to APRS-IS.