richonguzman/LoRa_APRS_iGate #238: APRS-IS filter activation

iuliandj, 2025-01-22:
Hello,

Can you activate, please, the next filters in IGate firmware to be defined without time and distance ?

b/call1/call2/call3/call4 ...
e/call1/call2 ...
d/digi1/digi2 ...

Original filter without additional filter in the back, like thy are defined here  https://www.aprs-is.net/javAPRSFilter.aspx,

The stations with call1/call2 etc. can not be heard directly on RF by IGate, so that is why they must be pull out from IS to local RF by IGate.

Thank you!

richonguzman, 2025-01-22:
I guess you are understanding this wrong.

The filter is to listen from APRS-IS all messages from the defined filters you have linked.

the problem is you wanting to Tx to RF packets that are more than 100kms away. I see not purpose of it if they can't be heard directly or with digirepeaters in between.

maybe you want to see what its happening on another part of the world, so Tx to RF packets from there too?

SQ2CPA, 2025-01-22:
`The stations with call1/call2 etc. can not be heard directly on RF by IGate, so that is why they must be pull out from IS to local RF by IGate.` -- and thats because of...?

iuliandj, 2025-01-22:
I have my purpose to hear packets on RF that are from another part of the world. You can say so. 
I think those filters work on other aprs devices right ?

If you cand activate them also on lora board I will make sure that you will be rewarded for your work. You know what I mean.

richonguzman, 2025-01-22:
with the filters you propose you will receive from APRS-IS all the beacons of this far away stations.

all of this data is readable over Serial output or even TNC connection.

do you need this also be Tx into RF?

iuliandj, 2025-01-22:
Yes, the information from those filters must be sent to RF.

iuliandj, 2025-01-26:
In section APRS-IS of IGate firmware, plese also add a check "Gate APRS-IS Positions to RF", "Gate APRS-IS Telemetry to RF", "Gate APRS-IS Beacons to RF"
Thank you!