richonguzman/LoRa_APRS_iGate #240: OLED time out activation

iuliandj, 2025-01-24:
Hi,
Can you please activate in the next update of IGate firmware, OLED time out also when board is not connected to WiFi? 
I saw that when Config WiFi AP is on, and WiFi is not available, the display stay on even is set tu turn off after x seconds.

I need this because tomorrow I will install a Lora APRS Digipeater in a location with no WiFi available, so I activated Config WiFi AP mode all the time. To reduce energy consumption and protect oled lifetime I think that is a good ideea to turn off even with AP activated all the time. 

Thank you!

S57PNX, 2025-01-25:
Why do you want to keep Wifi AP on all the time? Wifi is a massive power consumer, much bigger than the OLED display that worries you. OLED consumes aroun 7mA while Wifi AP consumes round 70mA.

richonguzman, 2025-01-25:
Digi EcoMode is implemented to use 1/10 of the current 

Do you really need WiFi in your installation??

iuliandj, 2025-01-25:
Because I want to connect by AP at need. If AP is set to search at 10 min WiFi network, oled is still on all the time.

iuliandj, 2025-01-26:
Yesterday I installed the digipeater, and today I installed an WiFI extender to bring internet to the board.

RobertMavric, 2025-02-05:
I (when I need to) send commands via APRS over the internet and use "QUERIES" ;)
http://www.findu.com/cgi-bin/entermsg.cgi and
https://github.com/richonguzman/LoRa_APRS_iGate/wiki/104.-QUERIES
?APRSSEM or ?APRSEEM

richonguzman, 2025-02-05:
> I (when I need to) send commands via APRS over the internet and use "QUERIES" ;) http://www.findu.com/cgi-bin/entermsg.cgi and https://github.com/richonguzman/LoRa_APRS_iGate/wiki/104.-QUERIES ?APRSSEM or ?APRSEEM

i don't know to what this is related, but this are to activate or not EcoMode for digipeater mode without wifiAP and Oled Screen