richonguzman/LoRa_APRS_iGate #241: T-beam versions - AXP192 is replaced by an AXP2101. And sending temperature in degrees of Celsius.

Jan-OK2, 2025-01-26:
As in title: In version 1.1 sends tracker in beacon voltage and current (+- charging/discharging) and in V1.2 voltage and percentage of baterry. Information of current is more useful, because percentage is calculated from baterry voltage and it can be estimated from this. Is'nt possible to get info about current from this other MPU and send it? Thanks

And one request for all devices using WX sensor, mostly WX station: Is it possible to send (in comment? telemetry?) acurate temperature in degrees of Celsius readed from BME280? I know, that degrees of F is a part od APRS standard for WX beacon, but its too inacurace without decimals.
Thanks

richonguzman, 2025-01-26:
Axp2101 has no way to measure current (only axp192 did that)

As for Celsius : wx packet is in F. But is has enough resolution. Right?

Jan-OK2, 2025-01-27:
1/ OK. thank You for informatiom. I thought that newer means better...

2/ Resolution is the right word, not accuracy! :-) I like to watch RAW data or comment with degrees Celsius with decimal, readed from T sensor (BME280 in this case), not calculated integer F degree value (which is for APSR format necessary) and back calculated in C degree by aprs.fi. Maybe see for inspiration: https://aprs.fi/?c=raw&call=OK2ILA-6&limit=50&view=normal
It is my very old project with WX station, that sends via RS232 APRS WX frame with C in comment to an old OpenWRT router with script, which it reades and puts into APRS-IS. At the same time, it compares and records max and min values of the day and sends it as comment in next beacon with the time stamp when was recorded. 
MAX and MIN values since last reboot (e.g. "Reboot mode" active - 24hours) in comment maybe would be useful too...
But I thing it is too much work with programming and I don'nt know, how many users will appreciate it.

richonguzman, 2025-01-27:
I guess this comments of min and max would be super niche custom mod

but I'm revisiting this and also other process of the WX LoRa APRS station.

Jan-OK2, 2025-01-27:
Yes I understand

But some way, how to read  current temperature exactly in C would be great! (without re-calculate from integer F, e.g. in aprs.fi).

richonguzman, 2025-01-30:
> Yes I understand
> 
> But some way, how to read current temperature exactly in C would be great! (without re-calculate from integer F, e.g. in aprs.fi).

I guess it all depends on where or how you are seeing this info. Over Aprs.fi or similar webpages or decoding it form RF with a program?

Jan-OK2, 2025-01-30:
I am sorry, don't understand. If I send to APRS WX data, temeprature is in integer value of Fahrenheit degree, and aprs.fi it recalculate back to Celsius with one decimal place.  This recalculation and rounding generates error about 0.4-0.5 C. Accuracy of BME280 is +-0.5 C. So total error can be almost about 1C. It is my pleasure to see temperature as accurate as possible. So I want to see value readed from BME280 in comment (as voltage e.g.). Better would be telemetry, but it is too much packets and maybe difficult to create.

richonguzman, 2025-01-30:
> I am sorry, don't understand. If I send to APRS WX data, temeprature is in integer value of Fahrenheit degree, and aprs.fi it recalculate back to Celsius with one decimal place. This recalculation [...]

will try to edit and test the code of the WX packet info gathering.

i wont asume much super perfect Temperature from BME280 ... as the error is "ok" and its a PRESSURE measuring device with Temperature correction. But not a perfect Temperature module.

although I will try to revisit the calculations to have a little more robust data...

Jan-OK2, 2025-02-01:
Yes, I know, but this accuracy is for so cheap device sufficiently good.
I found in **/src/wx_utils.cpp line 221**: String tempStr = generateTempString(((newTemp + Config.wxsensor.temperatureCorrection) * 1.8) + 32);
It is formula for conversion C to F. Please forgive me unknowledge of C++, but I think that in "newTemp" is value readed from sensor (in next step corrected by user defined value). And my idea is to get this corrected value and convert it to ASCII string and put it into comment part of paket with one decimal place (e.g. -01,2C). Like now is Batt=X.XXV
But I don't know, how high resolution in "newTemp" is. If too low then it doesn't make sense...

richonguzman, 2025-09-09:
this is on the list for next update

closed 2025-09-09 by richonguzman