richonguzman/LoRa_APRS_iGate #244: One-Wire  e.g. DS18B20  Sensor  ?

dl4sdr1989, 2025-01-29:
Is it possible  adding   one-wire sensor e.g e.g. DS18B20? will use a waterresistant Temperature Probe
tnx any info
Herbert   dl4sdr

richonguzman, 2025-01-29:
Hi Herbert,

besides adding this sensor, what is the intended goal beside getting T° info?  send it as a comment? as Telemetry?

dl4sdr1989, 2025-01-30:
Hallo Ricardo,

if possible  Telemetry. 73

richonguzman, 2025-01-30:
Ok

But WX telemetry or generic telemetry?

dl4sdr1989, 2025-01-30:
GM  es  sri.......WX please....mni tnx

arizonajon1, 2025-03-21:
I'd prefer the temp sensor be part of general telemetry, as I have concerns about battery temperature. Ricardo, is there a way to allow the temperature to be selectable as wx or as general tlm? Also, I'd prefer temp in degrees C, since battery ratings are primarily specified in C

richonguzman, 2025-03-21:
I'm returning to see how to do this... I was thinking this could be used in a encoded telemetry packet instead of the regular WX packet.

Jan-OK2, 2025-03-23:
#241 Here I also wrote about the possibility to show the temperature in C in the comment (my wish). And there is also an issue ( I can't find it) where we discussed the modes of sending Igate beacon with connected BME280 sensor (beacon with WX data) and the voltage as telemetry packet at the same time (why not). And the possibility to send both voltage and temperature in one telemetry packet (Bme280 or DS18B20 - both send raw data in C)...

And one more question: Has anyone considered supporting SHT11 sensors? I know it doesn't use standard I2C, but the Sensibus protocol. But these sensors are also used and are quite accurate.