richonguzman/LoRa_APRS_iGate #245: Ethernet / POE

MILUX-DE, 2025-01-31:
Are there still plans to integrate some of the known DIY Ethernet solutions (e.g. OE5BPA) based on WT32-ETH1, LilyGO T-ETH series, W5500 or TTGO T-Internet PoE boards? I've seen this issue mentioned several times in the past, but I can't figure out the actual condition.

richonguzman, 2025-01-31:
Hi

Yes , I hope to add this eventually

I found a way to test w5500 and maybe make it run

Will test next week

As for the OE5BPA board I got offered many times to receive one by different HAM OPs but never happened

manitwo10, 2025-02-03:
OE5BPA is what I also struggle with, LoRa and LCD can be wired to external header. Ethernet with PoE onboard. The HAT board was mostly sold in EU, shipping to SA will be expensive. I would sponsor 50% of such base board if you can manage getting it youself from fare east. https://de.aliexpress.com/item/4001134936204.html?

dl1lmh, 2025-02-04:
You could incorporate #192. Then you would already have a board + drivers

The AddOn board is in my repository and anyone can recreate it. 

If you need the addon board to test, I can send it to you...

petrkr, 2025-02-11:
Check my fork https://github.com/petrkr/LoRa_APRS_iGate/tree/lansupport

I have there support for OctopusLab board with LAN Shield, which supports PoE. I have it deployed like that.

[image]

petrkr, 2025-06-22:
I see yoy used same names like me, so it will conflict, why you  not rebased it and not used my half year work?

petrkr, 2025-06-23:
Also I not doing PR of not done work, thats why it was in branch, I needed some changes from @richonguzman those days, dnk if all are merged, especially move to sdk 3.x

petrkr, 2025-06-23:
https://github.com/richonguzman/LoRa_APRS_iGate/pull/142

Here is first mention about it, september..

UT2UH, 2025-06-23:
I sent PR only for the core and libs upgrade to support my board. Not for Ethernet support. What is my best move on your opinion now?

petrkr, 2025-06-23:
> I sent PR only for the core and libs upgrade to support my board. Not for Ethernet support. What is my best move on your opinion now?

To be universal across boards. Which i wanted achive, but if this PR will be merged my whole work will be messed up, because you use same names of files just for one board.

But do not worry. I will not continue, it make sence and I do not want waste my time anymore for this project if they do not want my work, why to waste it?

petrkr, 2025-06-23:
That is just reference as evidence...

Anyway this is NOT closed and I posted branch ar Feb too

https://github.com/richonguzman/LoRa_APRS_iGate/issues/245#issuecomment-2651363447

And pls, do not talk about  Ukrainians, we have already enough them at our pub, already more than 4 times ran away without payment and making lot of mess there, acting as pigs, always get some chips for ber/vodka and then thise chips are everywhere...
Also one bar here must be closed by city hall, because police came there almost everynight...

But that is not part of this PR/Issue

richonguzman, 2025-06-23:
Anyone taking about beer???

Hi all! I tested many months ago to update libraries and all I could from @petrkr , but could not achieve to have a smooth update

So it was on hold , and it's good that we could update it all if you help me

UT2UH, 2025-06-24:
@richonguzman, ready to help but if you upgrade the version without merging disrupted PRs they are not on hold but rather abandoned. If these PRs included new boards - the only cure from collaborators frustration will be the GPL license. There should be some policies for the painless collaboration - PRs for boards and new features are separated, PR to support new boards come first, if the new board requires library upgrades such upgrades come earlier and so on... dev branch?.. expected in the next release list?.. till GPL do us part... )

S57PNX, 2026-01-03:
I have a Lilygo T-ETH-ELite + 433 LoRa Shield on my desk and I am trying to make it work. I've got the basic pinout, GPS, LoRa radio and and currently spending time  to get W5500 ethernet running - and then I found this thread. 

I can see that both @petrkr and @UT2UH have done a lot of work in this area, and I would hate to duplicate my efforts to what was already done. Can we get some of their solutions accepted? 

To start, adding T-ETH-ELite as a new variant is a quick task, we just need a board_pinout.h and platformio.ini (UT2UH has a better board_pinout.h than my experiments). This will make this platform work perfectly over Wifi.

Then, adding Ethernet support is a separate and bigger effort, simply because there is no ethernet in the project yet. This will need a few more new files (eth_utils.c and .h), some changes to existing source logic and to the index.html too. Maybe even library updates? This can be done in smaller consecutive steps.  After ethernet is in and running for one board, it is again relatively easy to add other platforms with ethernet ports, like Olimex, OE5BPA, OctopusLAN and others.

Thoughts?

S57PNX, 2026-01-03:
@petrkr  thanks for the feedback. After some research, I see now that the SDK version  and Platformio is a big and sensitive topic,  and since I am not part of this project, the decision if/when switch to pioarduino remains with @richonguzman . Also, my availability to work on anything nontrivial will expire with the end of holidays, so staring next week $dayjob will take over again.

For now I'd be happy if at least the T-ETH-ELite variant goes in, so the board works at least with wifi. The whole issue of adding Ethernet is almost certainly beyond my skillset and experience.

petrkr, 2026-01-11:
> Hello,
> 
> I have ESP32-POE from Olimex [https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware ](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware) and I was trying implement Ethernet support but wihout success. Tried https://github.com/dl1lmh/LoRa_APRS_iGate_for_OLIMEX but with this solution radio is not working. I have radio SX1276 (E32 900M20S) so 900MHz but I ordered radio SX1262 (E22 400M30S). My version of radio is working with variant ESP32_DIY_LoRa_915 together with WiFi from richonguzman github.

As you using legacy PHY LAN7820 you should be able use stock old SDK with my hard-coded Ethernet support.

> The idea is implement iGate solution on the roof close to antenna via only one ethernet cable to have power and Internet.

Yes that was I was about that too.. But I still need few pull requests to be merged,,, but meanwhile I say they are conflicting again and now I working on different projects (for example remote SSTV transmission for some contents - https://github.com/petrkr/rigremote or library for Tropic square's secure element, which could have use in HAM radio too), so I do not have time for this one :(

petrkr, 2026-01-12:
@damso  @S57PNX 

updated lansupport Proof-of-concept to latest released version now. I added detection of 4.x and 5.x SDK in Ethernet class, so it should be able to build with old 4.x SDK and also with new 5.x SDK (but there are some new problems with MQTT Client, because it also use old WiFiClient from SDK4.x)

Lot of things to do in order to properly support Ethernet (or SPI one, RMII ethernet works already in SDK 4.x)

damso, 2026-01-12:
Thank you @petrkr 
With my current radio I have only success with variant ESP32_DIY_LoRa_915 so with your branch I was trying also this to be sure that Lora radio started.

First of all I modified the GPIO pin for Ethernet:
```
ETH.begin(ETH_PHY_LAN8720, 0, 23, 18, 12, ETH_CLOCK_GPIO17_OUT);
ETH.begin(0, 12, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_LAN8720, ETH_CLOCK_GPIO17_OUT);
```
because I had error: 
_E (7244) lan87xx: lan87xx_init(499): wrong chip ID
E (7245) esp_eth: esp_eth_driver_install(215): init phy failed
[  7279][E][ETH.cpp:324] begin(): esp_eth_driver_install failed_

For first test I was using: Compiling ETH init: **SDK 4.x (Arduino core 2.x)**") and LAN is working for a while (2-3pings) and connection is lost…

[image]

Next I tested: ("Compiling ETH init: **SDK 5.x (Arduino core 3.x)**") and I did some test and it works without problem! Ethernet works, radio works.

[image]

but If I see correctly with your version, WiFi is disabled permanently so if I disconnect LAN cable WiFi will not work?

damso, 2026-02-03:
Hello!

So I think you observed similar issue like me: "For first test I was using: Compiling ETH init: SDK 4.x (Arduino core 2.x)") and LAN is working for a while (2-3pings) and connection is lost…" but I didnt notice that it is related to LoRa TX.

I will check your code with my board. Thanks @petrkr

petrkr, 2026-02-03:
btw @damso 
latest lansupport moved PIN definitons to board_pinout.h and requires -DHAS_ETH to platformio.ini for that board to be enabled.

inspiration here

https://github.com/petrkr/LoRa_APRS_iGate/blob/lansupport/variants/ESP32_OctopusLab_LAN_LoRa/board_pinout.h

https://github.com/petrkr/LoRa_APRS_iGate/blob/lansupport/variants/ESP32_OctopusLab_LAN_LoRa/platformio.ini

damso, 2026-02-18:
Problem with i2c was my fault of GPIO assignment. With SDK 5.x (Arduino core 3.x) I didnt notice any problems.