richonguzman/LoRa_APRS_iGate #246: igate 433mhz v2025.01.22 on Heltec v3 doesn't send information to aprs from other trackers/stations.

temmink, 2025-02-07:
I have installed Igate version 2025.01.22 on a Heltec v3 device, for both TX and RX on 433Mhz.

The igate unit is using manual entered gps coordinates because it doesn't have an GPS, is connected to my wifi.
Configured options are: 

Enable APRS-is Connection, 
Gate APRS-IS Messages to RF    -Gate APRS-IS Objects to RF).

I send beacon via APRS-IS and RF, with 15 minutes interval.

The problem is that I also have 433mhz tracker (lilygo TTG, with GPS and the CA2RXU tracker firmware).
I see that the igate is receiving the tracker information (and vice versa).
It also looks if the Igate is sending the information from the tracker to the aprs network. (Indication on the serial interface).

**Connecting to APRS-IS ...     Connected!
(Server: euro.aprs2.net / Port: 14580)
-- Sending Beacon to APRSIS --
-- Sending Beacon to RF --
---> LoRa Packet Tx    : ABLSTR-05>APLRG1,WIDE1-1:!L4'wOO],;a  GLoRa APRS GW - Blue**

When I press the button on the tracker I get the following information on the the console of the igate:

**<--- LoRa Packet Rx    : ABLSTR-7>APLRT1,WIDE1-1:!/4'uoO],s[%,G
(RSSI:-42 / SNR:7.00 / FreqErr:554)
---> Uploaded to APRS-IS**

But I cannot see the tracker on maps, also when I checked live data from stations in my area, I don't see the trackers data re-send by the igate.
However I can see the beacon (only from the igate) received by the aprs network.

**[APRS] 07/02/2025 19:12:16+01:00: [ABLSTR-05](https://aprs.to/station/11799881)>APLRG1,TCPIP*,qAC,T2CSNGRAD:!L4'wOO],;a GLoRa APRS GW - Blue (Type: Position)**

So is there something wrong with the setup, or is there an issue with the igate software?

richonguzman, 2025-02-07:
Hi,

It seems your callsing does not follow any country am-radio rules

ABLSTR??

temmink, 2025-02-07:
you could be right, but the login is validated.  (Igate passed by passcode, and the  tracker doesn't have a passcode to setup.
Could that be the issue?

richonguzman, 2025-02-07:
Passcode could be hacked to make it work. But the code has a validation for callsign.

Could you test your real callsign?

temmink, 2025-02-07:
I can try and see if that will make a difference tonight. 

But because I'm using the 433Mhz band, which is not necessary to be licensed when you transmit with low power, this shouldn't matter in my area. When I'm using my own call-sign there are other limitations in place, like it's discouraged to use the call-sign when communicate with non-licensed users of the network. But I can do this to check if it solves my problem.

Will let you know! 

BTW: you have a fast response time! ;-)

richonguzman, 2025-02-07:
> BTW: you have a fast response time! ;-)

sometimes, as I pass my time playing with code in the PC , hahaha

test for just RX packet (not Tx) and this way the license won't be a problem you should face with your country rules.

temmink, 2025-02-07:
unfortunately, it doesn't make a difference. I also changed the tracker call-sign, but this doesn't make a differences.

Will now test with only RX activated. Give me a moment , hahah

richonguzman, 2025-02-07:
> unfortunately, it doesn't make a difference. I also changed the tracker call-sign, but this doesn't make a differences.
> 
> Will now test with only RX activated. Give me a moment , hahah

test valid Callsign, with Rx , and passcode calculated to this valid callsign ;)

temmink, 2025-02-07:
Test with valid Callsign, with only RX options on the igate (igate has valid login, status on aprs network).
I can see the 'new' igate appearing on the map (and live packages on the APRS network), but no information from my tracker is send by the igate.  In the console log of the Igate, I only see the following line:

**RX / GPX CALL_SIGN-7 / APLRT1 / WIDE1-1 / -37dBM / 7.00dB / 513 Hz / longitude / latitude /   0.1km**

nothing else.

richonguzman, 2025-02-07:
Does your tracker appear on aprs.fi?

temmink, 2025-02-07:
No, it doesn't. 
It only got displayed once, because somebody else uploaded it to the APRS.fi (I think yesterday).
But it has never being uploaded by my Igate.

richonguzman, 2025-02-07:
write me on telegram (same github user) !

I'm curious of what going on there as it should go right away

richonguzman, 2025-02-08:
great news it finally worked for you

closed 2025-02-08 by richonguzman