richonguzman/LoRa_APRS_iGate #247: Barometer altitude from GPS

texasyojimbo, 2025-02-10:
Just a thought -- if we have GPS running, it would be nice if we had an option to get the BMP280 barometer correction altitude directly from the GPS altitude.

The current static value for height correction is great for static igates, but if someone has a mobile igate, they'll either have to accept that the pressure is "wrong" (perhaps set to zero meters above MSL) or constantly change the value.

richonguzman, 2025-02-10:
hi,

I get it, but "a moving Wx station" is normally not a good Wx station data.... 

will try the fix for this scenario ;)

richonguzman, 2025-02-25:
now , when board has gps and gps is enabled in igate , height correction will be used from gps altitude data

closed 2025-02-25 by richonguzman