richonguzman/LoRa_APRS_iGate #254: Voltage telemetry sending - what's the conditions?

Jan-OK2, 2025-02-21:
I would like to monitor the repeater's power supply using telemetry (solar power monitoring), but I don't know what the conditions are for telemetry to work in the iGate settings. It doesn't work for me in any combination. When I turn it on in config, it stops sending the voltage (Batt/Ext) as text in the beacon, but I don't see it as telemetry on aprs.fi.
I have tried beacon combinations RF/APRS-IS, Voltage Batt/Ext - On/Off etc, but it still only goes to send text as a comment.

richonguzman, 2025-02-21:
hi,

first I must gather some info:

igate is sending Wx Packet? where you sending battery info as comment text? or as encoded telemetry?

Jan-OK2, 2025-02-21:
OK, your query gave me a little hint - I guess I can't send WX and voltage telemetry at the same time (?). This is what parts of my packets look like on aprs.fi (Voltage Int and Ext are both still ON):

1/ Send voltage as telemetry ON and at the same time WX as telemetry ON:
**.../...g...t037h59b10331Experimental LoRa Repeater+WX**
(no voltage info at all)

2/ Send voltage as telemetry OFF and at the same time send WX as telemetry ON:
**.../...g...t037h60b10331Experimental LoRa Repeater+WX Batt=4.19V Ext=0.88V**
(voltage info only in the comment)

3/Send voltage as telemetry ON and at the same time send WX as telemetry OFF:
**:EQNS.0,0.01,0,0,0.02,0
:UNIT.VDC,VDC
:PARM.V_Batt,V_Ext
:!L58y^S "vW_ GExperimental LoRa Repeater+WX|)j%^!Q|**
(Telemetry seems OK, but no WX data)

Why is it not possible to send both? "WX telemetry" is not true telemetry, but a special human readable string of data in one packet. I understand the large number of packets for telemetry, but WX is just 1 extra...

richonguzman, 2025-02-21:
aprs-is packet decoding dont process encoded telemetry on a Wx packet. Cant do it. I asked the same way : "WHY?" and answers was, can't do it.

so is one or the other.

Jan-OK2, 2025-02-21:
I'm not sure we understand each other. But if that's not possible, I respect that. WX data and voltage as a comment in one packet is also useful...
But my point was that a WX packet is a single special packet with all APRS attributes (incl. comment), which also contains WX data in APRS WX format. But telemetry data is sent as a set of several packets with definition and values of different variables. So a WX packet is not the same as telemetry packet and can be sent separately. Why not?: 1 WX beacon packet, 4 telemetry packets? That's how I thought the setup worked.
I understand it's not good for APRS traffic, but sometimes it can be useful. On aprs.fi, for example, you can see nice graphs of battery voltage in sunlight and air temperature at the same time.

richonguzman, 2025-02-22:
I got you now. You want a packet with Wx data and another as a "pure" telemetry packet to send battery data and few more things eventually, right?

Jan-OK2, 2025-02-22:
Please don't get me wrong (maybe my poor English, sorry). This was not a request, but a question about how it works (see title). How does what combination of checkboxes work. I think I understand it well now. I don't want to waste your time on unnecessary things, I was just wondering technically - WX telemetry is just different telemetry than others. Thanks for such a great project and Your answers!

richonguzman, 2025-02-22:
but its still a good idea to take care of it eventually

closed 2025-02-25 by richonguzman