richonguzman/LoRa_APRS_iGate #261: Add option to enable wifi AP with a button when in Eco mode

5B4APV, 2025-03-01:
Hello, I think it will be nice to have the option of enabling the wifi AP with a button again when the device is in Eco mode, so that it can be configured/updated without the need of reflashing it.
A switch could be used that directly enables - disables the Eco mode and reboots the device (directly linked to the option in the web UI)

richonguzman, 2025-03-02:
hi @5B4APV if you send a query/message at the Digipeater Station it can start, stop EcoMode. When EcoMode is stopped it will reboot and WiFiAP will be enable for you to do any mods to the configuration.

Check the Wiki in the Query Commands available.

5B4APV, 2025-03-03:
Thanks for the reply, I didn’t notice that possibility. I just tested it and it seems to work.
There might be a few instances when a button would be the only option, but they should be rare.

closed 2025-03-18 by richonguzman