richonguzman/LoRa_APRS_iGate #263: Blacklist not working as expected

pd9mwo, 2025-03-02:
Blacklist not working as expected

when adding a call sign to the blacklist as specified it does NOT block it.
I found the solution but this is not as intended.

First add callsign to black list
example : AB2CDE* HJ4WSE-5 XY6Z*
press save and reboot

then **disable** all these options : 
_enable aprs-is connection
gate aprs-is message to rf
gate aprs-is objects to rf
enable lora TX
enable lora RX_
Yes you read it right ! DISABLE THEM
press SAVE and reboot

allmost done ... **enable** these options again .. so **ENABLE** : 
_enable aprs-is connection
gate aprs-is message to rf
gate aprs-is objects to rf
enable lora TX
enable lora RX_
press SAVE and reboot

Now youre black list is working
Please fix this , many thanks in advance

73

richonguzman, 2025-03-02:
hi @PA9MWO , this is really strange as the writing process of saving the configuration to internal memory has nothing to do with having or not aprs-is switches enabled or not.

when you save the data (without your special process) do they get saved into memory? (do they appear after reboot written in the blackList box?

PA9MWO, 2025-03-03:
hello  Ricardo 
Yes ... sometime its getting written sometimes it doesn't.
I cleaned my Lilygo Lora32 1.6 completly ( with esp tool ) and reflashed it. 
clean config installed.
I left it running this night and this morning i see the black listed has  not picked up.  
The item i put in the black list is not working.

There is a guy in my neighbourhood that is not a HAM radio guy and is illegaly using a fake callsign. :-(
I just want to block / black list him

i hopes this helps.
73

PA9MWO, 2025-03-03:
Ok  !  I also  filtered the (illegal) callsign  in the APRS-IS filter. Thats ok for now.
But as i am a huge fan of sharing my info can you add a blacklist of filter in the syslog part.
Yes i want to share all my recieved stations with _lora.link9.net_ but not the illegal ones.

can you please add a filter in the syslog part ? 
Many thanks in advance
73 pa9mwo

richonguzman, 2025-03-03:
the blacklist process does block the "blacklisted" stations to be repeated and uploaded into aprs-is.

will have to check if this also blocks them to be sended into syslog.

beside this , where do you see the black list not working? showing them as received packets?

richonguzman, 2025-03-03:
now I get this problem

will add this blacklist method to avoid showing it on Rx Packets and on Syslog packets too.

PA9MWO, 2025-03-03:
Youre the best ! Thanks man !
73

richonguzman, 2025-03-03:
Please download the code from github (not on webflasher still) to test it

it should block over the hole process any blacklisted station:
no digipeater packet
no upload to aprisa
not shown on Rx packets
not logged over Syslog

PA9MWO, 2025-03-03:
I will test and keep you posted !
.. thx in advance

PA9MWO, 2025-03-03:
YES! great work !
the illegal stations are no longer available in the recieved packets and _https://lora-aprs.live/_ webpage.
many many thanks !
**! issue solved !**
73

richonguzman, 2025-03-03:
great, adding this to new release then ;)

closed 2025-03-03 by richonguzman