richonguzman/LoRa_APRS_iGate #270: LoRa iGate Received Packets refresh intermittent or not populating

arizonajon1, 2025-03-21:
I use the "Last 10 received packets list" tab on the LoRa iGate web server, specifically the build for the LilyGO T3 V1.6.1 (433 and 915), to attempt to monitor what the device is hearing. I've noticed over the past few months that while it states on the page that the "List refresh automatically every 15 seconds." in reality the refresh on that screen is really intermittent, and ultimately just won't refresh, or the refresh list is empty. The workaround that sometimes behaves is to click on "Configuration" or another tab then re-click on "Received packets". Doing that will sometimes restart the refresh, but other times does not. I've tried on Windows 11 using Chrome and Firefox, same behavior. On iOS, specifically on my iPhone, the behavior is the same on Chrome or Safari.
Certainly it's possible that the device hasn't heard a new packet in the past 15 seconds, but it's surprising to me when it shows the past 10 packets received then the next refresh shows zero packets, as I'd expect it to show the same past 10 packets.
Has anyone else encountered this issue?

Jan-OK2, 2025-03-23:
I have the same problem. One piece of the board shows received packets only a moment after boot and after some time the list remains permanently empty, although igate works normally.

richonguzman, 2025-03-23:
> I have the same problem. One piece of the board shows received packets only a moment after boot and after some time the list remains permanently empty, although igate works normally.

which station? to see the packets its uploading receiving?

Jan-OK2, 2025-03-23:
Just packets that igate heard. But it's a non-constant problem, I don't know what it depends on. Sometimes I see them. Otherwise it works normally - it repeats the packets, they're just not in the list. But it's not an important issue to me, just that it's been mentioned here.

arizonajon1, 2025-04-07:
Hi there - didn't know if anything has changed here. I've poked through the code but don't really know what I'm looking for $%(
The web interface is very important to me to show traffic when I'm out at a remote digi site and it's impractical to see the screen (like it's inside a sealed enclosure, or super sunny out). At home here after flashing a V1.6.1 device, the "log" available on the serial interface continues to show new traffic but the web interface just stops updating. In fact, some times it never even starts populating. Like I've said, I get this issue on Windows and iOS, in several different browsers on each.

[image]

richonguzman, 2026-02-17:
Hi Jon, I did not get a better way to show packets for now... but will try to reform all to be a better GUI for this