richonguzman/LoRa_APRS_iGate #271: Can't seem to connect to Wifi that doesn't use password

memm74, 2025-03-28:
I can't seem to be able to connect to a Wifi that is not using a password. When I leave the password empty in the web interface it doesn’t want to connect. I can connect to other Wifis though, i.e. Wifi networks that need a password.

richonguzman, 2025-03-28:
we can test a way if you can edit some code of the repository and test there locally uploading with VSCODE instead of the web flasher

sm3sgp, 2025-03-28:
On the same subject (wifi password) I got a report that wifi connect with a password containing the character "=" was not able to connect. But, I have not verified this myself (sorry).

richonguzman, 2025-03-28:
> On the same subject (wifi password) I got a report that wifi connect with a password containing the character "=" was not able to connect. But, I have not verified this myself (sorry).

as far as I checked, the "=" character should not be a problem for esp32 to connect

but if it was there should be plenty of combinations of regular letters to be secure enough.

closed 2026-09-30 by richonguzman