richonguzman/LoRa_APRS_iGate #275: Change Request: GMT Time Correction Option

snag1es, 2025-04-11:
Hi,
In another project, the POSIX time zone database is used, which simplifies the switch to daylight saving time in the summer and the return to standard time in the winter. Since most countries still use this system, would it be possible to arrange for this adjustment, along with adding the time server source address (IP or hostname)?

richonguzman, 2025-04-11:
Hi,
Can you explain the 3 topics a little more?

This proposed idea is to have "auto" change for time zone data in each season?

And the other two?

snag1es, 2025-04-11:
Hi,
Yes, the time does change—for example, from CET to CEST during the summer. Information related to the POSIX time zone data can be found here:
https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

The second point is to have an option to add a time server to synchronize the time, using an IP address or hostname such as es.pool.ntp.org, for example.

This idea is already implemented in the Meshtastic project, and perhaps some parts of their code could be used as inspiration. I'm not a developer—just thinking it might be possible.
Here is the relevant configuration section:
https://meshtastic.org/docs/configuration/radio/network/#ntp-server

There is no 3 ;)

richonguzman, 2025-04-11:
Will check about the changing time zone (does this take care of some countries changing outside the normal change date (sooner or later)?)??

As for NTP, this is already implemented in the code and showed ok the Rx received Packets on the WebUi

arizonajon1, 2025-04-11:
and here in Arizona USA, we don't participate in the ritual of changing our clocks. %^) So, whatever method might be implemented would also need to accommodate that.

snag1es, 2025-04-14:
The idea is to set a proper timezone, such as Europe/Paris, US/Arizona, or any other relevant timezone. Additionally, once the correct timezone is set, the system should synchronize time with a local time server, such as 0.arizona.pool.ntp.org.

This will automatically sync the date and time of the gateway when the module is connected to the Internet.

In short, the Time Zone parameter ensures the correct local time. If there is a time shift, the timezone adjusts the date accordingly. If there is no shift, no action is required. In case a leap second is declared for the current year, the Time Zone (TZ) handles it on the specified day. The time server provides the correct signal to synchronize the clock based on world time servers.

This is valid only for the Gates without GPS which is cheaper then installing GPS module on a fixed location.

richonguzman, 2025-04-14:
this could be solved by using ntp with known timezones defined, right? or are all ntp servers UTC defined?

snag1es, 2025-04-16:
Hi Ricardo,

No, I think the time server did not provide TZ data information - just the sync protocol. TZ data has been provided as a Linux package containing TZ information and handling the switching of daylight saving time.