richonguzman/LoRa_APRS_iGate #285: feature request - received Packets display of the date

wanderkarte, 2025-05-14:
First of all, thank you for the interesting project. Tracker and also iGate are running ufb. Thanks for the work, time and effort! I am using the current version 3.2.
In the section “Received Packets” the last 10 packets are displayed with time. Is it possible to display the date as well?
Since my iGate is located in a not very active APRS area, the date would be quite interesting, at least for me.

richonguzman, 2025-05-14:
hi, this info could be also shown , but maybe for your use-case it would be better to enable the syslog into the recommended "free" syslog and see there all the received beacons from a few days or month back

wanderkarte, 2025-05-14:
Thanks for the quick reply. Of course. Syslog. I completely forgot about that. That's the way. Many thanks for the tip!

closed 2025-05-14 by wanderkarte