richonguzman/LoRa_APRS_iGate #288: alternative WX data source

JimlyTheReal, 2025-05-25:
Hi OM Ricardo, 

the iGate is located in my shack. 
However, the weather data should come from outside the house. Laying a 10m cable to implement I2C is neither possible nor sensible for me. 

If I install an ESP32 with a BME280 or BMP280 as a bridge, which could send the WX data to the iGate via WLAN UDP, that would be a viable solution. 

What do you think of this suggestion? 
Could you implement an alternative UDP reception? 

This would require specifying a UDP port. The WX data would then come, for example, either as a CSV string or as JSON.

73 de OM Jimly

w5nmr, 2025-05-30:
A very good idea
MQTT would also be conceivable

richonguzman, 2025-05-30:
i have already a few updates on MQTT coming soon (still learning a few tricks) maybe we could add this as a way to sold this issue too :)

greghall-au, 2025-07-23:
https://github.com/richonguzman/LoRa_APRS_Wx_Station is a very good Wx.
You could build one of those ;)
Or maybe there is scope for RS485 from the iGate to sensors outside?