richonguzman/LoRa_APRS_iGate #289: Feature request: NTP server customization

dscp46, 2025-05-31:
Hi,

It would be nice if we could customize, on the admin web interface, the NTP server IP address/FQDN.

The purpose is to reduce reliance on external resources on hardened sites (with a local NTP server that relies on a local GPS receiver and a backup DCF77/ALS162 decoder).

Best regards,

richonguzman, 2025-06-01:
ok, but this gets really away from "other" NTP servers

is the goal of this request to have a box to input a personal choice NTP server to start?

bergie, 2025-10-01:
This would be quite useful also for our sailboat use case. We have internet connectivity only every once in a while, and the rest the device could operate as a digipeater. We do have a local NTP server on board to keep all devices in sync.