richonguzman/LoRa_APRS_iGate #290: Impossible to change a map icon

m0ppr, 2025-06-01:
Hi, I cannot change icon - I try to change from blac circle into red diamond from drop down menu, and then I press save changes, device roboots and after rebooting it is again a black circle.

richonguzman, 2025-06-01:
there is no black circle, maybe black diamond?

what other configurations changes did you do on the igate WebConfiguration page?

maybe something is blocking the saving part of the code

m0ppr, 2025-06-01:
Hi, I can change description and save it with no problem. Sorry, I was wrong, it is a blue (not black) circle and I no chance save it. It switches to anything else when I select any different option from drop down menu, but after saving and restarting it is back to the previous icon. If I change description and save it after restart it is saved.

richonguzman, 2025-06-01:
there was not picture on the message... please try again

richonguzman, 2025-06-01:
so all is working fine with Red Diamond, but as soon as you save with new symbol it gets not saved? that's the only problem?

richonguzman, 2025-06-01:
this is really strange as nothing is blocking the selection of the symbols in the code at all

are you using google chrome web-browser? did you refresh the web-page once it gets back connected? sometimes its a cache thing on the browser

richonguzman, 2025-06-01:
i like to find out what's going on with your selection

(selection should be saved no matter what on the configuration before re-booting)

m0ppr, 2025-06-01:
Ok, thank you. I did not want to waste your time as I fought it is just on my device. So how can I start diagnosing?

S57PNX, 2025-07-20:
This is intentional. aprs.fi will only diplay your WX info if your symbol is a blue L (it is a decision that they made). So Ricardo's firmware will automatically switch to a blue L if your WX sensor is switched on in the configuration.

m0ppr, 2025-07-20:
I had no clue about it, thank you!!!

richonguzman, 2025-09-06:
Hi John,

this is exactly a proposal i want to push soon:
have telemetry with regular symbols but not as Wx Station... to have another way to see data over time even though you don't want it to be recognized as a Wx Data Station. Regular Telemetry will be added as a possibility .

Right now to make aprs.fi decode the Wx Packet you need to have the blue circle + L running.