richonguzman/LoRa_APRS_iGate #308: Automatic power on

Uberseer, 2025-07-29:
I plan to install my LoRa APRS module in the trunk of my vehicle, turning on with vehicle ignition power.  However, I realize that once the unit has powered off (e.g. from low battery condition), once power is restored it does not turn on until a button is pressed.  (I'm using a Heltec v3)

Is it possible to make the unit automatically power on through firmware?  Or is this purely a hardware issue?

richonguzman, 2025-07-29:
this boards have no hardware to detect a connected voltage and do something. They have a chip that enables charging the battery connected but this works outside the esp32 possibilities.

so the only way would be to add some resistors and transistors to have somehow a hardware reset switch for this

Uberseer, 2025-07-29:
Oh I will pursue this from a hardware standpoint.  Thanks

SQ2CPA, 2025-07-30:
You don't need to detect connected voltage, you can just deep sleep for like 5 minutes and then check voltage on startup and then sleep again or turn on normally. 

If this board doesn't have low power protection built-in then it is recommended to install that outside the board if you don't have that built-in in your battery (because some li-ions have built-in protections). Software protection is always not perfect and such physical protection board (which costs very little) will protect your cells

Uberseer, 2025-08-10:
In a nutshell the problem is that I expect the board to power on (boot) when USB power is applied.  I contacted Heltec who told me this board does automatically power on when USB power is applied, and if not then it's the firmware being written to the board (APRS Tracker).

I'm a bit confused...I didn't think the firmware would be responsible for commanding the board to start on power up.  But maybe I'm wrong.  Does this makes sense?

I can confirm that when selecting POWER OFF from the menu, the board doe NOT power off.  So maybe there is some firmware issue?

closed 2025-09-09 by richonguzman