richonguzman/LoRa_APRS_iGate #309: Igate Will not connect to wifi extender

zaech, 2025-07-30:
Using a Netgear EX6120 wifi extender to reach to a somewhat remote igate with the following config:
TTGO_T_BEAM_V1_2
Igate Version date: 2025-06-20
Build date: 2025-06-19 16:16:27 UTC

the igate will not connect to the extender (other devices have no issue connecting) and always and uses the second SSID in the Wifi Access list.
The extender SSID and password are different than the router . I have tried Address reservation and normal DHCP for the igate in the router, both produce the same result.

Is there something unique about connecting to an extender the the igate is not supplying or is the SSID connection period too short?

Thanks,
N2VEN

richonguzman, 2025-07-30:
does the extender has " " or special characters in the SSID or Password?

zaech, 2025-07-30:
SSID is: ext_zach and password contains only letters and numbers (11 characters long). The second SSID (which it does connect to) is: zach_net and password has only letters and numbers (9 character long)

richonguzman, 2025-07-30:
can you avoid the "_" and test it on the repeater SSID ? (i've seen some esp32 had trouble with them on SSID and check that the SSID and PASSWORD of the first one do not have any " " before or after the characters too

zaech, 2025-07-30:
Removed the _ in the SSID no change. Confirmed there are no " "  before and or after the  SSID or password

zaech, 2025-07-30:
2.4G and 5G capable, I am using the 2.4G path to the repeater (selected by use of the SSID). I have also tried turning off the 5G path in the repeater, no change.

richonguzman, 2025-07-30:
how many WIFI are defined and saved on the board?

save just the needed one and delete all the others (also use the needed one in the first position = don't have a empty wifi defined at first)

zaech, 2025-07-31:
Only two are defined, first one is the repeater, second is the primary wifi router. 
If the first and only one is the repeater, the igate goes into AP mode (192.168.4.1)

richonguzman, 2025-07-31:
with other firmwares can you connect to the repeater? any other firmware with wifi?

zaech, 2025-07-31:
I can connect Android Phone and Tablets. I don't have any other IOT type devices to try.

zaech, 2025-07-31:
I have loaded Meshtastic FW and it also will not connect to the repeater! Must be something in the ESP32 wifi code that is not handling things properly, your thoughts?

richonguzman, 2025-07-31:
> I have loaded Meshtastic FW and it also will not connect to the repeater! Must be something in the ESP32 wifi code that is not handling things properly, your thoughts?

show/paste here a picture of your board.

zaech, 2025-07-31:
[image]

richonguzman, 2025-07-31:
> I agree, I'm out of ideas also that's why I sought your help.! I am going to look into purchasing a different chipset board, is there one you could recommend?

different LoRa board? or different WiFi-Repeater ?

zaech, 2025-07-31:
Different LoRa board

zaech, 2025-07-31:
I configured the repeater in AP mode, the t-beam can now connect to it. Although not ideal for my application (since an ethernet connection is now needed) I can move forward.
Still a mystery as to why repeater mode will not work.

richonguzman, 2025-07-31:
so at least your board is not bad! :)