richonguzman/LoRa_APRS_iGate #311: 433 MHz Heltec V3 Output Power

arizonajon1, 2025-08-13:
Hi Ricardo - I just got a Heltec V3 up and running, and it measures +19.7 dBm output. I'm running on the USB connection. If I use the battery connection, will it get to +22 dBm? Do you know? It seems according to the Semtech literature that +22 dBm is available directly from the battery supply?

richonguzman, 2025-08-13:
you should get 22dBm +-1 dBm as per Heltec Datasheet.

i don't know if it would change from battery or usb but i guess it should be the same.

how are you measuring it? many measuring units can't measure bellow 1W.

richonguzman, 2025-08-13:
some values are not available with the used library to control lora modules (RadioLib)

i guess 19 and 21 are not available at all.

what does it show for 20 and 22 on sx1262  boards?

richonguzman, 2025-08-13:
i did some "bellow the desk" corrections:

sx1278 get till 20
sx1262 if set as 20 do 22
.... i will have to fix this to get it free for each type of board

arizonajon1, 2025-08-13:
hmmm. I am using the Heltec V3 board, as far as I know it is an SX1262. The LilyGO units with the SX1278 would not make more than +17 dBm without the "high power" bit flipped so this board definitely has more output power.
The equipment has been on for several hours now so should be relatively stable.
Currently powered via USB.
The left column is the value I sett using the web configuration page for TX output power, the right value is the measured power.

+16 = +15.6
+17 = +16.5
+18 = +17.5
+19 = +18.2
+20 = +18.9

Now powered via battery connector with supply at 4.2 Vdc
+16 = +15.6
+17 = +16.5
+18 = +17.4
+19 = +18.3
+20 = +19.0

so there's no difference between USB and direct via the battery connector.

Next is to see what is the voltage at the SX1262 Vbat pin.

arizonajon1, 2025-08-14:
I cannot enter a value that is greater than +20 dBm. I get the same error. I prolly could hack the firmware.

N1IOX, 2025-08-14:
I was noticing the same thing with the Heltec V3.2 915 MHz board (supposedly capable of +22 dBm) - the web UI will not allow a value greater than 20.

Just don't do what I did to try to work around it: change the live, in-browser 'lora.power' field's 'max' attribute to 22.  It sure let me apply it, but then the module wouldn't boot up ;)  I needed to re-flash to recover, which of course was quick / easy.

richonguzman, 2025-08-14:
just to be clear: max is defined ass 20

but this means: on boards with SX1278 is set up as 20. On boards with SX1262 it adds (internally)  + 2, so this is really 22dBm.

i guess i will have to clear this out on the setup/config webpage to have the user selected value more transparent

richonguzman, 2025-09-09:
this has been added for new release list of TODO