richonguzman/LoRa_APRS_iGate #314: T-DECK PLUS NO KEYBOARD ERROR

arizonajon1, 2025-08-31:
This has been an issue ever since i first set up my T-Deck Plus, but I cannot write a message. If I go to MESSAGES>WRITE I get the error "no keyboard detected - can't write message" 
Anyone else seen this? I'm using the tracker software from March this year, not the most current.
Cheers and 73 - Jon N7UV

richonguzman, 2025-08-31:
The most current version solved this !

arizonajon1, 2025-08-31:
my friend KA6ZDM just installed the most current and he can't get the keyboard to work. Let me see if I can get him over here.

1nerdherder, 2025-08-31:
Hello - I downloaded the 3.0.1 August version from the web flasher to a LilyGo Tdeck plus with the new keyboard - and it is not seeing the keyboard.  Could it be the new style (oval soft key) keyboard vs. the old (blackberry like) one?