richonguzman/LoRa_APRS_iGate #322: Feature Request: RAK13800 WisBlock Ethernet Module support

Uberseer, 2025-09-10:
I would like to request support for the RAK13800 Ethernet module (based on the W5100S-L chip from WIZnet).  That would allow the RAK4630 boards to be used as iGates.

Details of the chip can be found here: https://docs.rakwireless.com/product-categories/wisblock/rak13800/overview/   As well, Meshtastic already supports this chip so open source code is available to accelerate the project.

(I apologize if this is a duplicate - I thought I created an issue a while back but can't find it)

richonguzman, 2025-09-10:
hi!

this has not been forgotten at all! Its just that i was finishing the update of the iGate V3.1

will go into this very soon to start with it (do you have already the board with you?)

ocgltd, 2025-12-30:
Any update on this?  I have 2 RAK boards with the RAK13800 module using POE, waiting for this feature.  This board is ideal to stick up on a repeater tower and use power over ethernet!!  

If this feature isn't going to happen I may have to buy different boards.  Hopefully this feature is on the todo list for 2026!

richonguzman, 2025-12-30:
Hi,

i'm on it, i hope to have a valid testing firmware this Friday (without the holidays in between for you to test )