richonguzman/LoRa_APRS_iGate #324: Possible new WebUI configuration bug in version 25-09-09:

Jan-OK2, 2025-09-13:
When I activate Reboot mode, any further changes to the settings are repeatedly blocked. This makes it impossible to use Reboot mode. It took me a long time to figure out why I suddenly couldn't save the settings before I found the connection to this. 
Tested on TTGO V 2.1. and Heltec wireless stick lite (more boards of both)..

N1IOX, 2025-09-13:
I was deploying another node today (Heltec 32 V3 433MHz) with V3.1 (2025-09-09), and also had issues saving changes.  Though in my case, I was trying to set initial params after flashing a device for the first time, with Reboot mode still in its default state (off).

I ended up reflashing that node with V3.0.1 to get it deployed quickly.

Ricardo - I have a couple more nodes currently unused, and would be happy to perform any tests to help debug this with them.

Jan-OK2, 2025-09-14:
During further testing, I encountered random problems with saving (=not saving) the configuration even when Auto reboot was turned off. But when it was turned on, they occurred always .

The behavior was the same regardless of whether the configuration was set after a clean webUI flash, a flash via VSC, or from a saved .json file.

richonguzman, 2025-09-14:
could you send me over telegram a copy of the .json file of the configuration? I suspect we are reaching the size of the file where we need to increase the size on the firmware

Jan-OK2, 2025-09-14:
Hello.

I don't use Telegram. This is a test settings for Heltec Wireless Stick Lite.

Jan, OK2ILA

 

Tento e-mail byl antivirovým softwarem Avast zkontrolován na viry.
www.avast.com

richonguzman, 2025-09-15:
lets hope more test this to have this issue solved :P

> I noticed the V3.1.1 release, with the JSON size fix. I deployed that to a spare node, and now see no issues saving the config.
> 
> Thank you, Ricardo!

Leo-PL, 2025-09-16:
This is probably the same issue as I mentioned in discussions: https://github.com/richonguzman/LoRa_APRS_iGate/discussions/304 - but still no joy with 3.1.1. 

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f7977  PS      : 0x00060830  A0      : 0x800e1e6d  A1      : 0x3ffdf8e0  
A2      : 0x00000010  A3      : 0x3f42091d  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffe0df0  A7      : 0x3ffe0df0  A8      : 0x800eb590  A9      : 0x3ffdf8c0  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000009  A13     : 0x3ffe37b4  
A14     : 0x00000012  A15     : 0x3ffbc208  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001f  LBEG    : 0x40092361  LEND    : 0x40092371  LCOUNT  : 0xfffffffd  

Backtrace: 0x400f7974:0x3ffdf8e0 0x400e1e6a:0x3ffdf900 0x4019fdc3:0x3ffdf970 0x400eec89:0x3ffdf990 0x400ecd0d:0x3ffdf9e0 0x400ecd55:0x3ffdfa20 0x4018c5bd:0x3ffdfa40 0x4018c635:0x3ffdfa80 0x4018cf62:0x3ffdfaa0

ELF file SHA256: cae01a029331faa0

```

richonguzman, 2025-09-17:
available now!

Leo-PL, 2025-09-18:
Works for me. Thanks!

richonguzman, 2025-09-18:
> Works for me. Thanks!

great

lets wait for @N1IOX and @Jan-OK2

Jan-OK2, 2025-09-19:
Tested on TTGO V2.1. It seems to be OK! 
Tomorrow, I will test the Heltec board and I believe that the problem will be solved.

richonguzman, 2025-09-19:
> Tested on TTGO V2.1. It seems to be OK! Tomorrow, I will test the Heltec board and I believe that the problem will be solved.

please write back when you do