richonguzman/LoRa_APRS_iGate #327: Send altitude in an APRS-standard format

darksidelemm, 2025-09-20:
The RSGB appears to have flown this code on a balloon launch today - https://aprs.fi/#!call=a%2FGB1HAB&timerange=3600&tail=3600

Unfortunately, the packet format only has altitude in the comment field. This means that software that parses APRS packets will not find the altitude information. That includes aprs.fi, and sondehub-amateur.

The APRS spec does support sending altitude as part of the compressed position information - adding this would allow software to extract the altitude information correctly...

darksidelemm, 2025-09-20:
My apologies - I wasn't aware this software could be configured to send altitude as part of the base91 packet - could you point me to where that happens?

richonguzman, 2025-09-20:
base91 GPS encoding payload use the 3 last bytes/characters to send Speed+Course or Altitude. The last byte shows what data is it and the first and send has the data to be used

richonguzman, 2025-09-20:
> Sure - what I meant was - does this software support that right now (e.g. could that balloon launch group have configured it to do so), and if so, how to enable it?

i offer help to do any changes to the code. But was told it was not needed.

richonguzman, 2025-09-20:
> Ah. I guess they might not have understood the implications of only having the altitude sent in the comment field. I guess they know what the issues are now!
> 
> Anyway, it would probably be a useful addition if anyone else wanted to try and use this software on a high altitude balloon launch, or just wanted the ability to send their altitude...

i will add a pure balloon FW soon.