richonguzman/LoRa_APRS_iGate #328: Issue reporting negative elevations

arizonajon1, 2025-09-26:
SE California has the Salton Sink, which drops more than 60 meters below sea level. In driving back and forth between Phoenix and San Diego, one regularly passes into below sea level elevation. However, the display on the T-Deck is unhappy with negative elevations, as shown in the photo here

[image]

I haven't gone through the code to figure out what causes this weird elevation/altitude value, but figured perhaps Ricardo would know what needs to be done and where to do it.

Cheers and 73 - Jon N7UV

arizonajon1, 2025-09-26:
No, I was thinking more of just showing it as "A=-009m". nothing fancy.

richonguzman, 2025-09-29:
now altitude works with negative values (doesn't add any "0" before... )

check also that LoRa settings are fully customizable now... all ranges of CR4, BW and SF

richonguzman, 2025-09-29:
> Muchas gracias Ricardo! i just looked at the Web flashing tool - there doesn't seem to be a notification of a new version (dated today) available. Is this something I should download into platform.io and flash on my own?

this is available only for Github + VSCODE updates for testing before

once i gather many updates and they run well i put them into a new release for webflasher

arizonajon1, 2025-09-30:
Just an update on real-world testing.
Yesterday I set up a BW62.5 SF7 CR6 digi/igate using a qrp-labs light solar v1.0 and a mobile using my TTGO T-Deck Plus. I am running the M1/M2/M3 mode nomenclature which is configured to use BW62.5 SF7 CR6. Both units use the SX1262 which has the high-stability oscillator. It seems to work quite well even at 13 km distance (was getting -89 to -86 dBm RSSI). 
Soon, I'll drive back to SoCal and pass below sea level to check how the elevation parameter displays.

richonguzman, 2025-09-30:
good news... once I finish a few other things i will add the M1,M2,M3 mods for sure  (too)!