richonguzman/LoRa_APRS_iGate #331: Possible leading zero bug for Callsign designations.

Magnetic-Core-memory, 2025-09-26:
There may  be a leading zero bug for Callsign designations.  

I have had an issue that prevented the iGate processing received packets correctly and proved tricky to identify.
 
Superficuially verything appearsto be correctly function  however RF packets are not heard.

If you define a callsign to be Callsign-01  rather than Callsign-1 this will result in  iGate behaving oddly and not appearing to receive RF packets over the air.  Changing  from a -01  designation to a  -1 designation fully corrects the issue.  The callsign structure requires validation on entry.

Therefore leading zeros should probably be  prohibited by the callsign input code, or alternatively, permitted and handled correctly within the code.  
I only tested -00, -01, -02, -03 and -05, all failed.  -10 -11 etc.  are OK.

I am using the latest iGate software with a Heltek Wireless Tracker board.

Thanks.

arizonajon1, 2025-09-30:
I will note however that the way trad ax.25 does stuff the SSID is a 4-bit nibble, with values from 0 to 15 decimal (0-F hex). I don't know how the SSID is parsed; it the parser expects the potential of a leading "1" (as in -10, -11... -15) but it doesn't anticipate a leading "0" then there could be an issue. 

We really need to rethink how LoRa does the addressing and SSIDs. It's time to move beyond the constraints of old ax.25.

Cheers and 73 - Jon N7UV

richonguzman, 2025-09-30:
SSID should be a numerical character:

03 is not numerical, but 3 is.