richonguzman/LoRa_APRS_iGate #333: Propose to change LoRa configurations from Country Abbreviation to "MODE"

arizonajon1, 2025-09-29:
I apologize up front: this is not an issue. However, it's something that I think needs updating in the code for the tracker and I'm not familiar enough with "pull" requests to work this through that method. Hopefully I figure that out soon!

Overview:

Change the tracker "configuration" names "EU", "UK", "PL" to "M1", "M2", "M3". 
- The current names not only are confusing for operators who live in other than these three countries/regions.
- Some new to this firmware have asked me whether the radio configuration is somehow configured to meet the indicated country's regulatory environment. AFAIK there is no correlation.
- Having a specific configuration for a local network should not require the operator to remember it as "the Poland one", "the UK one", or the "European Union one".

As an example, my T-Deck Plus is currently set up with three "modes"
- M1: 433.775MHz, BW125, SF7, CR4:6 (operational APRS network deployed across 2500 sq km of the greater Phoenix area)
- M2: 433.775MHz, BW62.5, SF7, CR4:6 (experimental APRS network I've just set up here in Phoenix)
- M3: 433.775MHz, BW125, SF9, CR4:7 (when I visit San Diego and want to use their LoRa APRS network)

While I enjoy visiting the EU, UK, or PL, these are arbitrary names for a tracker's three available configuration sets each of which contains:
- Operating center frequency
- LoRa bandwidth
- LoRa spreading factor
- LoRa code rate

I have long familiarity with Motorola two-way radios and their enduring use of the term "MODE" to represent a container that holds a set of parameters that configure the radio for a given situation:
- Operating frequency (RX)
- Operating frequency (TX)
- CTCSS TX
- CTCSS RX
- CTCSS TX/RX
- Alphanumeric for display
- Conventional/Trunking
and many other radio configuration parameters specific to a situation or area.

My proposal requires only some text editing tweaks in three files: lora_utils.cpp; menu_utils.cpp, display.cpp

lora_utils.cpp
Change lines 87-89 to:
            case 0: loraCountryFreq = "M1"; break;
            case 1: loraCountryFreq = "M2"; break;
            case 2: loraCountryFreq = "M3"; break;

menu_utils.cpp
Change lines 381-383 to:
                    case 0: freqChangeWarning = "      M1 --> M2"; break;
                    case 1: freqChangeWarning = "      M2 --> M3"; break;
                    case 2: freqChangeWarning = "      M3 --> M1"; break;

display.cpp
Change lines 557-559 to:
        case 0: workingFreq += "M1]"; break;
        case 1: workingFreq += "M2]"; break;
        case 2: workingFreq += "M3]"; break;

I haven't found any other dependencies as of yet.

What do y'all think?

Cheers and 73 - Jon N7UV

richonguzman, 2025-09-30:
You are absolutely right

Will update it

richonguzman, 2025-10-03:
Having 4 modes:

GL
PL
UK
US

is also a good idea

wb2osz, 2025-10-03:
>  Could the global mode work for US as well (honest question) ?

That is a very good question.

In the US, the allowed frequency range for LoRa is 902-928 MHz.
This is an unlicensed ISM band so there is all sorts of random stuff there.
This range is also available for ham radio use, so some are using LoRa APRS here.
It is easy to find LoRa devices with that frequency range, often preloaded with Meshtastic.
It is difficult to find 433 MHz LoRa devices from domestic suppliers.
I don't recall seeing a proposal for a standard frequency for this band.

Other US hams are using the global standard frequency.
433 MHz falls inside the 70cm ham band so using LoRa here, under ham rules (callsign identification, no encryption, higher power, higher duty cycle) should be OK.
The problem is that digital data modes are limited to 100 kHz bandwidth, in the 70cm band, so the standard 125 should not be used.

The FCC (US government agency regulating radio transmissions) is considering changes to bandwidth regulations for the 70 cm amateur radio band, including relaxing the current 100 kHz limit. These efforts are part of a broader initiative to modernize outdated symbol rate restrictions across multiple bands.

73,
John WB2OSZ

arizonajon1, 2025-10-03:
Thanks for weighing in John!

The bandwidth limitation is one of the reasons I have continued to encourage Ricardo to allow at least the 62.5 kHz setting... Now it's time to move the local network to that narrower BW.

Cheers and 73 - Jon N7UV

s56oa, 2025-10-04:
I would love not to mix LoRA APRS with Meshtastic, Meshcore and similar systems.
Let's keep our beloved amateur radio frequencies clean and let's Meshtastic and Meshcore guys use their respective LoRA non-amateur radio frequency allocations.

I do play with both Meshtastic and Meshcore, but on 868 MHz EU non-amateur radio frequency allocation. 
It is like a CB radio. Fun, but many times completely un-controllable from human side of things. :)

Personally i do not mind "mode1, mode2, mode3" settings. But it would be good to keep extra text for end users not to "get lost" and still use correct settings.

Maybe a drop-down in web configurator for each mode: "EU", "UK", "PL", "Custom". And when user selects "Custom" he/she can enter custom settings for that LoRA mode.

This would still enable simplicity for settings and switching between different modes either for testing purposes and/or when roaming between geographical territories with different settings.